AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #714: Request for a more meaningful light color on the Magsafe charge connector

Danby108, 2022-07-05:
I'd like to see the little LED light on the Magsafe connector continue to show a meaningful indication of whether my MacBook has reached a full charge. For all my decades of owning Mac laptops, I've always thought that orange means my laptop is charging, while green means it's fully charged. As I interpret things, if I'm using your app to target an 80% charge, then it ought to show green when it reaches that point, because that's the current definition of what "fully charged" means.

I can't count how many times I've been reassured to see this little light come on, and to know a little bit more about what's going on with my computer. With my latest M1 MacBook Pro, there are no LED indicators on the side of the computer showing the charge state (and also the charging state), making that last remaining LED all the more important. And yet, as things are currently implemented, it only shows green when the battery reaches a 100% full charge state. Among other universal meanings, green generally implies that all is safe, and yet a 100% charge is not such a long-term safe state for the battery to be in!

For intel MacBooks, the Charge Limiter app says that it "modifies a parameter called BFCL ("Battery Final Charge Level") which controls the MagSafe LED indicator light to display the correct status." I don't know what the equivalent would need to be on M1 MacBooks, but this seems like a good idea and a worthy thing to do, for ALL types of MacBooks with MagSafe connectors. 

In thinking this through a bit more, it's clear to me that orange can't always indicate whether the battery is charging, because there are discharging times during calibration mode, etc., but green CAN always indicate that it's reached (or exceeded) it's targeted charge setting, or it's still within tolerances of being at that target (in Sailing Mode). If I've just pressed the "top up" button, I guess "fully charged" would temporarily mean 100%, so that the light would show orange again until it reaches that temporarily requested level.

Implementing this properly would take a bit of doing, yet this is a natural and meaningful bit of always-visible user feedback that your app can (and I believe should) take control over, as a helpful indicator of what's going on with the battery, regardless of whether the menu bar icon or anything else is currently visible.

Thanks for a very helpful app, and thanks for considering adding this when you can, to make the app a little bit better.

MatthiasKerbl, 2023-01-01:
Hi @Danby108,

We have just released AlDente 1.20 with a new beta feature called "Control MagSafe LED" which will finally properly indicate if the Charge Limit is reached. You can read more about this feature on our blog here: https://apphousekitchen.com/feature-explanation-control-magsafe-led/

Cheers,
Matthias

MatthiasKerbl, 2023-01-12:
Hi @Danby108,

Since we have now implemented the requested feature, I will close this issue.

Cheers,
Matthias

closed 2023-01-12 by MatthiasKerbl