AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #717: Paypal

ashutekale, 2022-07-10:
Please enable Paypal for purchasing as I would not enter my credit card details everywhere.

MatthiasKerbl, 2022-07-11:
Hi @ashutekale,

Paypal should be enabled and when I try to buy AlDente Pro, I can choose Paypal as well.
Where are you located? Maybe it is not possible to use Paypal in every country.

Cheers,
Matthias

ashutekale, 2022-07-21:
@MatthiasKerbl India, kindly request you to please enable so I can buy it ASAP.

MatthiasKerbl, 2022-07-21:
> @MatthiasKerbl India, kindly request you to please enable so I can buy it ASAP. Apple Pay is not available in India.
> 
> [image]

Hi @ashutekale,

You can use a Mastercard or Visacard to pay. Since all of our payments are managed by our payment provider Paddle, I will forward your request to them.

Cheers,
Matthias

closed 2022-07-21 by MatthiasKerbl

ashutekale, 2022-07-21:
I want to use Paypal please, I do not want to put my card details wherever required. Can you please enable it?

MatthiasKerbl, 2022-07-21:
> I want to use Paypal please, I do not want to put my card details wherever required. Can you please enable it?

Hi @ashutekale,

I am sorry but I can not do more than tell our payment provider Paddle.com to enable PayPal in India as well. Otherwise, you can try a VPN.

Cheers,
Matthias