AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #728: Which version ?

liquidsmok3, 2022-07-18:
Why does the app report its version number as 2.2 in settings and when you check for updates, but the GitHub for this app says the most recent version is 1.17.1 ?

liquidsmok3, 2022-07-19:
just in case I dont like the new version, is it still possible to download the "classic' version before i replace it.

MatthiasKerbl, 2022-07-21:
> just in case I dont like the new version, is it still possible to download the "classic' version before i replace it.

Yes, you can always go back to the old version here: https://github.com/davidwernhart/AlDente/releases

Cheers,
Matthias

closed 2022-07-21 by MatthiasKerbl

liquidsmok3, 2022-07-31:
I think im going to go back to the classic version.  This version doesn't seem to work as well.  My Mac has been sitting on 100% for 3 days which is the main point this app is supposed to prevent.  Also mostly all the settings are greyed out.  The older version would discharge but now I have to manually unplug the power to get it to discharge.  No clamshell support which again is one of the main points of using an app like this.  I mean I get it, you need to monetize it some kind of way, but why even have a free version that mostly doesn't do anything.