AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #733: Bug: Charging is completely disabled after sleep

Rob-ot, 2022-07-21:
**Describe the bug**
Hello, thanks for making this app. I've been using it on my MacBook Pro 14" since I got it 8 months ago.

I'm running into an issue where, after going to sleep and waking up, my computer "sometimes" stops recognizing that a USB-C charger is plugged in - even when the AlDente charge limit is set above the current charge, and even if the charge limit is set to 100%.

In the bugged state, OSX doesn't play a sound when the cable is plugged in and the OSX battery menubar says "Power Source: Battery" and seems completely unaware that a cable is plugged in.

I've also tried all 3 USB-C ports and different chargers. I tried uninstalling AlDente+helper for a week and this problem never occurred so it does seem to be caused by AlDente.

Rebooting the computer fixes the issue. Setting the charge limit to 100% does not fix it.

**To Reproduce**
Steps to reproduce the behavior:
1. Set AlDente charge limit to 80%
2. Plug in USB-C and charge to limit
3. Close laptop lid to sleep computer for 8 hours
4. Open lid
5. Bug: the system is now running on battery, even after battery drains well below charge limit, and system doesn't recognize charger is plugged in when unplugging/plugging in charger

**Expected behavior**
Plugging in a charger when battery is below charge limit should charge battery to limit.

**MacBook (please complete the following information):**
 - MacBook model A2442
 - macOS version 12.4
 - AlDente Free
 - AlDente Version 1.17.1
 - List of all enabled/disabled settings in AlDente: Free version so everything disabled

**Additional context**
More than happy to provide any logs or more information, or run a debug build if it would assist. I was trying to test older AlDente versions but was unable to find a way to download them.

Thanks again!

fragster9, 2022-07-28:
I have this same issue since upgrading to 12.4, I hadn't narrowed down exactly what caused it but it worked perfectly prior to this, it sounds correct though that it's related to sleep. Rebooting was the only fix I found too besides removing AlDente completely.

axeljov, 2022-07-30:
I've had the same exact same issue on 12.4. Rebooting is the only fix.

emersoncloud, 2022-08-03:
I'm having this issue as well. Does anyone have a solution yet? I've had to stop using Aldente altogether

fragster9, 2022-08-03:
I've found it seems to be better on 12.5, not 100% sure it's resolved but seems to be happening much less frequently than on 12.4 where I needed to reboot daily.

CrawX, 2022-09-28:
I'm also seeing this since a few days, running macOS 12.6 on a 2020 13" M1 MaBook Pro. Not sure what triggered this for the first time since I've upgraded to 12.6 when it came out... It happened two days in a row now. Will try to uninstall AlDente and see if it still happens then.

Rob-ot, 2022-09-28:
An update from me:
It hasn't done this in over a month now and I've been using 1.17.1 the while time. That points to some variable outside of AlDente.

MatthiasKerbl, 2022-11-09:
Hi everyone,

Since a lot of you are reporting this issue after a macOS update, it might be related to this known issue #768.
Try the following steps to fix it:

> AlDente stops working after the macOS update!
> 
> Unfortunately, AlDente temporarily does not work after a macOS update for a small number of users.
> 
> To get AlDente to work again, please try the following steps which have worked for many users before:
> 
> 1. Unplug and discharge your MacBook to less than your current charge limit in AlDente
> 2. Turn off your MacBook and close the lid for a couple of minutes
> 3. Unplug the power brick from the outlet and wait a couple of minutes
> 4. Turn on your MacBook and plug it in again
> 5. Check if it stops charging at the charge limit set in AlDente
> 
> We are aware of this bug for a while now but were so far not able to fix it. However, we will keep trying.

Cheers,
Matthias

MatthiasKerbl, 2022-11-09:
> An update from me: It hasn't done this in over a month now and I've been using 1.17.1 the while time. That points to some variable outside of AlDente.

Any update from your side @Rob-ot or was AlDente working as intended?

Cheers,
Matthias

Rob-ot, 2022-11-09:
I haven't had this issue for over a month. After reading through #768, it appears to be the same issue. I'll close this issue. Thanks for the update!

closed 2022-11-09 by Rob-ot

Rob-ot, 2022-12-26:
One last followup to this issue, I believe it was (mostly?) caused by a defective USB-C cable I was using for charging, along with the Apple charging brick.

The cable+brick was able charge my Android phone fine but would sometimes fail to change the laptop.

I understand that USB-C charging is more complicated than connecting two wires, it's negotiating wattage and charge direction over the cable as well, and I believe this is where the issue happen. It was providing enough power to run the computer but not charge the battery.

After getting a new cable everything is working perfectly.

MatthiasKerbl, 2022-12-26:
> One last followup to this issue, I believe it was (mostly?) caused by a defective USB-C cable I was using for charging, along with the Apple charging brick.
>
> The cable+brick was able charge my [...]

Hi @Rob-ot,

Thanks for sharing.

Cheers,
Matthias