AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #735: Broken on macOS 12.5

bdkjones, 2022-07-23:
I updated macOS to version 12.5 a couple days ago and have discovered that Al Dente is no longer functional. Starting from a fully-charged state, when I set a desired charge level of 70% and click the "discharge" button, the lightning bolt disappears from the battery icon in the menubar, but then reappears a few seconds later. 

This process continues: the power source alternates between battery and power adaptor every few seconds. I am running Al Dente 1.17.1 and I have ensured that "optimized battery charging" is disabled in System Preferences. Rebooting my Mac has no effect. 

Has anyone else seen this behavior since updating to macOS 12.5?

bdkjones, 2022-07-23:
I suspect the cause of this may be that I have *two* external displays plugged into my 16" MacBook Pro via USB-C. Each of these displays is capable of acting as a power supply for the laptop.

When I reboot my Mac with just a single external display plugged in via USB-C (a Pro Display XDR), then click the "discharge" button in Al Dente, things work normally.

That said, I've been using the dual-display setup for months and have never had a problem with Al Dente previously. Perhaps something changed in macOS 12.5 that's causing the system to cycle from one "power adaptor" (external monitor 1) to the other "power adaptor" (external monitor 2) when Al Dente tells the system to run on battery to discharge?

zemtt, 2022-07-25:
I have the same issue on macOS13 beta3.

AlDente doesn't work with neither USB-C charging or MagSafe charging.

andrewdcross, 2022-07-27:
Yeah same thing here, was going along just fine then I guess some interaction between the software and the OS kicked in a couple of hours ago. Will try a restart and see if that fixes anything.

andrewdcross, 2022-07-27:
A restart seems to have resolved it for now.

zemtt, 2022-07-28:
> Yeah same thing here, was going along just fine then I guess some interaction between the software and the OS kicked in a couple of hours ago. Will try a restart and see if that fixes anything.
> 
> EDIT: 12.4, for reference.

Yeah! Fixed by a restart.

dayvista, 2022-07-29:
I'm experiencing the same issue. The charging state toggles on and off rapidly. It started around when I updated to 12.4 or 12.5, I can't remember.

Restarting fixes it temporarily. The problem usually comes back after a half hour or so. It appears to even be missing with normal charging when AlDente is turned off.

andrewdcross, 2022-07-29:
> Yeah! Fixed by a restart.

Yep, it's been stable for me since that restart. Now hopefully bdkjones has had some better luck with it...

fragster9, 2022-07-29:
Hope that works for you but I've been on the latest version and having the same issues, can take a day or two to manifest after a reboot so be interested to hear if you find the same.

andrewcrook, 2022-08-18:
I haven't had any issues with Mac OS 12.4 or 12.5.
Just updated to macOS 12.5.1 and AlDente 1.18 this morning

andrewcrook, 2022-08-18:
> I have the same issue on macOS13 beta3.

Well macOS 13 is still in beta and AlDente doesn’t officially support it yet I believe.
So not surprised. Might not be the same cause.

hostrogo, 2022-08-19:
I  had an issue with AlDente 1.18 and macOS 12.5.1. The Mac hanged. I stay on 1.17.1. I must say I use Monterey and Al Dente on a machine which is not supported (MBP mid 2012), with OpenCore Patcher.

andrewcrook, 2022-08-19:
macOS 12.5.1 and AlDente 1.18 on M1 13” MBP 2020 seems to be working fine

serifreedom, 2022-08-23:
> macOS 12.5.1 and AlDente 1.18 on M1 13” MBP 2020 seems to be working fine
Yes, 1.18 works on M1 Pro/Max 2021, macOS 12.5.1, typeC charge
- Stop charging when sleeping
- Stop charging when app closed

bdkjones, 2022-09-15:
I've narrowed this down to multiple USB-C/Thunderbolt displays connected. The procedure to fix Al Dente in this case is:

1. Disconnect all displays except one external.
2. Reboot your Mac.
3. Open Al Dente and begin discharging (assuming you're at 100%).
4. Plug in the second display
5. Your Mac will stop running on battery and switch to the A/C power coming from the new display.
6. Re-engage Al Dente's discharge mode.

hostrogo, 2022-09-15:
> I had an issue with AlDente 1.18 and macOS 12.5.1. The Mac hanged. I stay on 1.17.1. I must say I use Monterey and Al Dente on a machine which is not supported (MBP mid 2012), with OpenCore Patcher.

Finally this bug had nothing to do with Al Dente, version 1.18 works fine.

andrewcrook, 2022-09-15:
> I've narrowed this down 

@bdkjones 
You never said what model of Mac you have. 
How the software works varies between different Mac models.
mainly with or without T2 chip and Intel or Apple Silicon.

bdkjones, 2022-09-15:
I'm on a 16" MacBook Pro with the M1 Pro chip. The issue manifested again with the last macOS update to 12.6.1

MatthiasKerbl, 2022-11-01:
Hi @bdkjones,

If I understand correctly, the issue was that AlDente didn't work anymore after the macOS update.
This is unfortunately a known bug which occurs for some users.
This is the current fix we provide in our FAQs:

> **AlDente stops working after the macOS update!**
> 
> Unfortunately, AlDente temporarily does not work after a macOS update for a small number of users.
> 
> To get AlDente to work again, please try the following steps which have worked for many users before:
> 
> 1. Unplug and discharge your MacBook to less than your current charge limit in AlDente
> 2. Turn off your MacBook and close the lid for a couple of minutes
> 3. Unplug the power brick from the outlet and wait a couple of minutes
> 4. Turn on your MacBook and plug it in again
> 5. Check if it stops charging at the charge limit set in AlDente
> 
> We are aware of this bug for a while now but were so far not able to fix it. However, we will keep trying.

Our official thread about this topic is #768 and I will post any updates about this bug there. Therefore, I will close this issue but let me know if you have any new findings and I will open it again.

Cheers,
Matthias

closed 2022-11-01 by MatthiasKerbl