AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #742: Online license validation every month, why?

tl87, 2022-07-30:
Hello.

I bought the AlDente Pro Lifetime License and it comes with a Lifetime License, however today i got a warning when i booted my Mac, that said: "AlDente needs to check if your license is valid once every 30 days. Please connect to the internet for AlDente to work."

Why is this necessary? I don't think it should be required, when you have bought a "Lifetime License".

tl87, 2022-08-06:
I really hope they will fix it soon?

MatthiasKerbl, 2023-01-30:
Hi @tl87,

In general, we are trying to keep a good balance between usability/convenience and copyright/piracy issues. We know that there is no single best solution, especially not for every one of our users. However, we are trying to find a good middle ground and might reassess when necessary.

> Why is this necessary? I don't think it should be required, when you have bought a "Lifetime License".

Unfortunately, it is necessary for copyright and piracy prevention reasons to validate even the lifetime license regularly.

> My biggest concern is; what happens when the server, that does the validation now longer exists? Then you have an application, that will only partly work, since all the "Pro" features will be disabled.

Currently, licensing is outsourced to a partner company that offers licensing services to thousands of companies with millions of users. Therefore, it is highly unlikely that these servers just "no longer exist". Even if anything would happen to our licensing partner, we would try to fix it as quickly as possible by switching to a different partner or doing it in-house.

> Please fix this and remove the continuously validation of a Pro Lifetime License. I think it's fair to validate the license once when the license is first used and that should be it.

Unfortunately, this is not possible due to copyright and piracy prevention reasons.

Hope that helps.

Cheers,
Matthias

MatthiasKerbl, 2023-01-30:
> I agree and have the same issue still, i reported this before here [#544 (comment)](https://github.com/davidwernhart/AlDente-Charge-Limiter/issues/544#issuecomment-1085510104) and it is still an issue.
> 
> Very annoying, since i purchased the software...

Hi @notDavid,

The mentioned bug from #544 got fixed a while ago and should not be an issue anymore.

Cheers,
Matthias

tl87, 2023-01-30:
@MatthiasKerbl 

Thanks for the reply.

I understand the issue of piracy and would like suggest a possible solution to the issue, if you haven't already thought of that??

When one purchase a Pro License and afterwards wants to activate it in the software on their Mac, then how about you pair the license with the hardware ID of the computer. When someone else try to activate the same lincense, but with another hardware ID, activation will fail. Only the person who purchased the license and has the correct hardware ID, can then use the software. If the person then get's a new Mac and wants to use the software on the new Mac, then the person would be required to contact you with a the license key and hardware ID, in order for the licesen to be able to activate on a new Mac.

I hope this makes sense? English is not my native language :)

MatthiasKerbl, 2023-02-01:
> @MatthiasKerbl
> 
> Thanks for the reply.
> 
> I understand the issue of piracy and would like suggest a possible solution to the issue, if you haven't already thought of that??
> 
> When one purchase a Pro License and afterwards wants to activate it in the software on their Mac, then how about you pair the license with the hardware ID of the computer. When someone else try to activate the same lincese, but with another hardware ID, activation will fail. Only the person who purchased the license and has the correct hardware ID, can then use the software. If the person then get's a new Mac and wants to use the software on the new Mac, then the person would be required to contact you with a the license key and hardware ID, in order for the licesen to be able to activate on a new Mac.
> 
> I hope this makes sense? English is not my native language :)

Hi @tl87,

Thanks for the suggestion. It sounds interesting and I have forwarded it to our licensing partner.

tl87, 2023-02-01:
Awesome

MatthiasKerbl, 2023-02-18:
Hi @notDavid,

> It's up to AlDente to work around this issue (other software does not have this issue.)

How would you suggest do work around this issue?

> I purchased the lifetime license and had to uninstall the app, so i'm very disappointed.

I am sorry to hear that you are disappointed with AlDente. Do you have any improvement suggestions? Why did you have to uninstall AlDente?

tl87, 2023-02-25:
I really wish they could simply remove the "phone-home-feature"

panzer-arc, 2023-11-17:
Was working fine for a good while for me, but now I haven't been able to use my pro license for several days. I think it's on Paddle's end. I contacted support but haven't heard anything yet. @notDavid, I think you're making some good points here.

MatthiasKerbl, 2026-10-06:
Hi everyone,

Since the license verification issues mentioned here have not been reported again in this thread for a long time, I will close this issue for now.

If you are experiencing license verification issues, please open a new issue and include a debug file so we can assess it better.

Best,
Matthias

closed 2026-10-06 by MatthiasKerbl