AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #743: I didn't get license after payment

yuanzhe-liu, 2022-07-31:
I have paid through alipay, the order number is src_1LRe6aL6R0cPTOVWpIGovSx9
my email is mechrapz@gmail.com, please resend the active code, thanks.

steeviee, 2022-08-02:
Me too. I just purchased this today and it has been several hours and I still have not received an email. Waiting....

yuanzhe-liu, 2022-08-02:
? can anyone at least review these issues and give me the active code

yuanzhe-liu, 2022-08-02:
@MatthiasKerbl

yuanzhe-liu, 2022-08-02:
@MatthiasKerbl

yuanzhe-liu, 2022-08-03:
3 days past, still no one gave the active code.
It's totally unfair, I suggest everyone who wants to buy this app, think it again.

DamonHD, 2022-08-03:
I'm sorry that you are having difficulties but (a) my payment worked fine and (b) have you considered for a moment that the authors may be on holiday or even ill so there is no need to start implying bad things about their character?

Rgds

Damon

yuanzhe-liu, 2022-08-03:
> I'm sorry that you are having difficulties but (a) my payment worked fine and (b) have you considered for a moment that the authors may be on holiday or even ill so there is no need to start implying bad things about their character?
> 
> Rgds
> 
> Damon

bro they are a company not some individual developer who makes a personal interested application just for curiosity. They made this for profit, so of course could I complain all this bad stuff. I'm customer, I paid for it. why should keep silent about this totally unfair thing?

DamonHD, 2022-08-03:
Precisely what they are so far as I can tell is two individuals doing this on their own in their spare time, largely because they are trying to do a good thing.  Their FIRST reaction is often to offer a refund.  They are not money-grabbing greedy guys.  This is peak holiday time in Europe and you've waited only 3 days before accusing them of theft, just about.

You could learn a little patience and politeness and how to deal with other human beings IMHO.

Report it yes?  Sling mud, no.

Rgds

Damon

MatthiasKerbl, 2022-08-03:
Hi @yuanzhe-liu,

I am sorry you are having trouble with our purchasing process.
I could find your purchase in our backend and the reason why you did not get the purchasing emails is that you spelled your email address incorrectly. I have forwarded the emails to the correct address now.

As described on our support page (https://apphousekitchen.com/support/), our GitHub page is not our official support and is only meant for our community to have a place to interact with each other. That is why we only go through new issues here occasionally.

Also, we have a precise description of what to do when you do not get the purchasing emails in the FAQ section of our pricing/purchasing website (https://apphousekitchen.com/pricing/):

"[I Did Not Get The License Code Or Any Receipt Emails After Buying AlDente Pro!](https://apphousekitchen.com/pricing/)
There are multiple reasons why you did not get your license code after the purchase:

1. Sometimes it takes up to 30 Minutes for the purchase emails to go through.
2. The purchase emails end up in the junk email folder.
3. You had a typo in the email address during your purchase.
4. Failures of the payment networks/providers.
Please contact the support team of our payment provider Paddle at [assist@paddle.com](mailto:assist@paddle.com) and include the email address you used to purchase AlDente Pro. The Paddle support will try to solve your issue as fast as possible."

Maybe we need to improve our website so that it is easier to find this solution.

Hope that helps.

Cheers,
Matthias

closed 2022-08-03 by MatthiasKerbl

yuanzhe-liu, 2022-08-03:
> Hi @yuanzhe-liu,
>
> I am sorry you are having trouble with our purchasing process. I could find your purchase in our backend and the reason why you did not get the purchasing emails is that you [...]

I don't know why I type my email address incorrectly since it was completed by my chrome and I actually didn't type anything there, which means theoretically it can't be incorrect as I only have one email cache right now.

I strongly advise you to improve the user experience, I DID read the FAQ and I DID have contacted with paddle but NO one ever replied me. After some waiting, in fact after sending 3 or 4 email without anyone reply me except a useless bot , I directly go to their twitter and reply my s*** experience there :-). (Of course the twitter had been deleted right now as they are a really company haha). Maybe only their platform users like your kind have access to contact a real person or some support service, whatever, but I can ensure you from customer side there is really little thing that I can do. 

I want to make this clear, it's not because I didn't get my license that makes me really angry, @DamonHD, and personally I believe I'm not a cruel guy to others. I will explain why I'm so politeness this time below. Meanwhile I won't admit I'm not patient, since from my experience, it's hardly to except a software issue being ignored over 12 hours. But I can apologize for this part if you and of course, these two developers, accept, besides my prejudice that they are a company looking for profit.
 
I'm also a program developer and I know sometime issues could happen out of everyone's expectation. But as a developer, we do have duty for delivering best user experience as we can, right? I don't know exactly who is responsible for my bad experience, maybe paddle platform since after all they are the software reseller, but I think there should be someone care about this, reply to me and fix the thing rightly in time. So the description like "You had a typo in the email address during your purchase." doesn't make any sense, especially when I followed the instruction to contact paddle and only found no one cared about this but meanwhile I knew you have reviewed some other issue and replied there [
](https://github.com/davidwernhart/AlDente/issues/745). It is this indifferent attitude makes me angry.

That's it.