AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #745: Sailing mode is not working

kevinhollmann, 2022-08-02:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
Sailing mode is not working

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model Apple M1 Pro
 - macOS version macOS Montery [12.5]
 - AlDente Pro
 - AlDente Version 1.17.1
 - List of all enabled/disabled settings in AlDente:
 - I have tried it with all settings

[image]
[image]
[image]
[image]

MatthiasKerbl, 2022-08-02:
Hi @kevinhollmann,

Did you read the feature explanation of the Sailing Mode to know how it works in detail? https://apphousekitchen.com/feature-explanation-sailing-mode/

Try the following steps to know if the Sailing Mode is working for you:

1. Set the Charge Limit to 80%
2. Set the Sailing Mode difference to 10%
3. Charge your MacBook until the Charge Limit
4. Unplug the MacBook and let it discharge to approximately 75%
5. Plug in the MacBook again and check if charging is paused due to the Sailing Mode

Cheers,
Matthias

MatthiasKerbl, 2022-11-01:
Hi @kevinhollmann,

Since you haven't responded in a while, I am assuming that Sailing Mode is working as intended. Therefore, I will close this issue for now.
If my assumption is wrong or you want to add anything, let me know and I will open the issue again.

Cheers,
Matthias

TremotinoVolante, 2023-07-11:
> Hi @kevinhollmann,
>
> Did you read the feature explanation of the Sailing Mode to know how it works in detail? https://apphousekitchen.com/feature-explanation-sailing-mode/
>
> Try the following steps to [...]

Is this also the behaviour when the macbook was shut down or put to sleep during #4 ? On my macbook al Dente always charges the battery to the charge limit bevore it turns into the state "sailing mode" 
Cheers Tremotino