AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #751: Reinstall helper application unavailable after canceling password dialog

retospect, 2022-08-07:
**Describe the bug**
Reinstall helper application unavailable after cancel.

**To Reproduce**
Steps to reproduce the behavior:
1. Click "Reinstall" button
2. Click OK in "Update Helper" popup
3. Click Cancel in the password dialog
4. Click "Reinstall" button again

**Expected behavior**
Process should repeat, ie, "Update Helper" popup should show again.
But nothing happens. Maybe forgot to dispose of Update Helper popup correctly?
Restarting app makes the button work again.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model M1 MBP
 - macOS version latest
 - AlDente Pro or AlDente Free? Pro
 - AlDente Version? 1.17.1
 - List of all enabled/disabled settings in AlDente:

MatthiasKerbl, 2022-08-10:
Hi @retospect,

Thank you for telling us about this bug. We will try to fix it with one of the next updates.

Cheers,
Matthias

MatthiasKerbl, 2023-07-28:
Hi @retospect,

This issue got fixed.

Best,
Matthias

closed 2023-07-28 by MatthiasKerbl