AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #756: AlDente never launches on login

bbyk, 2022-08-18:
I've been using this awesome tool for a few versions now and it appears it never starts automatically when I restart my M1 MBP 2021 (macOS Monterey). I have the `Launch at Login` option on. I typically notice that when I see the MagSafe cable plug light go green eventually and I am like yeah I must have forgotten to start AlDente

MatthiasKerbl, 2022-08-30:
Hi @bbyk,

Launch at login should work as intended on your model as well.
Did you try a complete reinstall of AlDente?

Cheers,
Matthias

DamonHD, 2022-08-30:
FWIW L-on-L doesn't seem to work for me either, on my MBA M1 12.5.1.  Else working nicely.

Rgds

Damon

MatthiasKerbl, 2022-09-01:
> > Launch at login should work as intended on your model as well.
> 
> Naturally. But it doesn't seem to.
> 
> > Did you try a complete reinstall of AlDente?
> 
> Yes, I did.
> 
> I also tried to un-toggle, toggle back the `Launch at Login` toggle that didn't work either
> 
> Is there any logging that I can enable to help troubleshoot the issue?

Hi @bbyk,

You could send us a debug file (https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).

Cheers,
Matthias

bbyk, 2022-09-02:
@MatthiasKerbl the feature seems to have started working after I did the following:

1. `brew remove --zap --cask aldente`
2. full reboot ( I didn't do this at first and went to #3 but immediately got caught into the infinite loop of a) helper application failed, please re-install -> b) install the helper -> c) go to the step a. )
3. `brew install --cask aldente`

If you like I can close the issue (and file again if it comes back) ?

MatthiasKerbl, 2022-09-06:
> @MatthiasKerbl the feature seems to have started working after I did the following:
>
> 1. `brew remove --zap --cask aldente`
> 2. full reboot ( I didn't do this at first and went to `#3` but immediately [...]

Hi @bbyk,

I am glad it works now.
Yes, please close this issue for now and come back if it happens again.

Cheers,
Matthias