AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #758: clamshell mode discharge doesn't work on m1 MacBook Pro. been using ardent since 1.01. but since the last few versions the discharge clamshell mode is not working . currently on the latest version

hardik746, 2022-08-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 - macOS version (https://support.apple.com/en-us/HT201260)
 - AlDente Pro or AlDente Free?
 - AlDente Version?
 - List of all enabled/disabled settings in AlDente:

**Additional context**
Add any other context about the problem here.

hardik746, 2022-08-30:
> Hi @hardik746,
> Could you give us more details about the bug?
> As a start, it would help if you could fill out the following questions:

Thanks you very much for replying, actually I saw that I had not updated my MacBook . After updating it the functions works as expected. Thank you David

MatthiasKerbl, 2022-08-30:
> Thanks you very much for replying, actually I saw that I had not updated my MacBook . After updating it the functions works as expected. Thank you David
> […](#)

That's good to hear. :)

Cheers,
Matthias

closed 2022-08-30 by MatthiasKerbl