AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #760: Aldente 1.18 not working in macos 13 beta

nimo-codes, 2022-08-23:
aldente 1.18 not workin properly in macos 13 beta but 1.17 was workin grEAT

MatthiasKerbl, 2022-08-30:
Hi @nimo-codes,

What exactly is the issue?

Cheers,
Matthias

mikenadeau, 2022-09-11:
I'm also experiencing issues with AlDente Pro under macOS 13 beta 7 (22A5342f). 2020 MacBook Air M1. Worked flawlessly on macOS 12 (latest).

It seems to work fine while the system is awake, but when it goes to sleep (not shutdown, just asleep), it will charge to 100%.

Here are my settings -

[image]

On a side note, I had to re-enable the heat protection limit and stop charging when app closed options, they seem to keep being disabled automatically/randomly. The stop charging when app closed option definitely doesn't work on macOS 13; the second I quit the app, my laptop starts charging to 100%.

I tried reinstalling the helper a few times - no luck.

I'm suspecting - 
- Low Power Mode - it comes On when on battery. Still didn't find how to disable it to test;
- Allow in the background - AlDente is allowed to run in the background, but if macOS is like iOS, this is disabled when low power mode is On.

I'm happy to test any beta version and report to you guys.
Thanks,
Mike

mrzv, 2022-10-26:
1.19.1 is not working for me in macos 13. It keeps cycling between paused and charging, all the way to and at 100%. Low Power Mode is set to Never and the Allow in Background is turned on. (I didn't need to change anything, it was already that way.)

MatthiasKerbl, 2022-10-30:
> I'm also experiencing issues with AlDente Pro under macOS 13 beta 7 (22A5342f). 2020 MacBook Air M1. Worked flawlessly on macOS 12 (latest).
> 
> It seems to work fine while the system is awake, but when it goes to sleep (not shutdown, just asleep), it will charge to 100%.
> 
> Here are my settings -
> 
> [image]
> 
> On a side note, I had to re-enable the heat protection limit and stop charging when app closed options, they seem to keep being disabled automatically/randomly. The stop charging when app closed option definitely doesn't work on macOS 13; the second I quit the app, my laptop starts charging to 100%.
> 
> I tried reinstalling the helper a few times - no luck.
> 
> I'm suspecting -
> 
> * Low Power Mode - it comes On when on battery. Still didn't try to disable it to test;
> * Allow in the background - AlDente is allowed to run in the background, but if macOS is like iOS, this is disabled when low power mode is On.
> 
> **UPDATE:** I disabled Low Power Mode, and made sure AlDente is allowed to run in background. I will test and report back!
> 
> **UPDATE 2:** after disabling Low Power Mode, everything seems to be fine -
> 
> [image]
> 
> [image]
> 
> I guess that the "allow in background" is a new thing in macOS 13, and that it's linked to the Low Power Mode, like it is on iOS. I will do more tests (and try to prove the theory) and report back!
> 
> I'm happy to test any beta version and report to you guys.
> 
> Thanks, Mike

Hi @mikenadeau,

Thank you for the detailed report.
For us, AlDente 1.19.1 is working as intended under macOS 13.
Do you still experience these issues with AlDente 1.19.1 under macOS 13?

Cheers,
Matthias

MatthiasKerbl, 2022-10-30:
> 1.19.1 is not working for me in macos 13. It keeps cycling between paused and charging, all the way to and at 100%. Low Power Mode is set to Never and the Allow in Background is turned on. (I didn't need to change anything, it was already that way.)

Hi @mrzv,

This might be a known issue for some people after a macOS update.
Here are the instructions how you should be able to get it to work again:

> Unfortunately, AlDente temporarily does not work after a macOS update for some users.

To get AlDente to work again, please try the following steps which have worked for many users before:
1.  Unplug and discharge your MacBook to less than your current charge limit in AlDente
2. Turn off your MacBook and close the lid for a couple of minutes
3. Unplug the power brick from the outlet and wait a couple of minutes
4. Turn on your MacBook and plug it in again
5. Check if it stops charging at the charge limit set in AlDente

We are aware of this bug for a while now but we’re so far not able to fix it. However, we will keep trying.

mikenadeau, 2022-10-30:
I've had it turned off for the past 3 weeks or so.

I'll test it with MacOS 13 RC/final in the next 2 weeks and report back!

Thanks!
Mike

MatthiasKerbl, 2022-11-14:
Hi @mikenadeau,

Any updates on this?

Cheers,
Matthias

MatthiasKerbl, 2022-12-24:
Hi @mikenadeau,

Since you have not reported back in a while, I will close this issue for now. However, feel free to get back and I will open the issue again.

Cheers, 
Matthias

closed 2022-12-24 by MatthiasKerbl