AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #761: Why does my mac doesn't charge unless I top up

naomipozo, 2022-08-23:
Hi! My battery doesn't charge unless I top up to 100%

cocobisc, 2022-08-24:
Same on me. I keep failed when I tried to re-plugin the charger, it's pretty annoying. I'm using 1.18 Pro version.

MatthiasKerbl, 2022-08-30:
Hi @naomipozo and @cocobisc,

could you please fill out the following questions and specify the problem you encounter?

Cheers,
Matthias

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 - macOS version (https://support.apple.com/en-us/HT201260)
 - AlDente Pro or AlDente Free?
 - AlDente Version?
 - List of all enabled/disabled settings in AlDente:

**Additional context**
Add any other context about the problem here.

closed 2022-10-04 by MatthiasKerbl