AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #764: Constant CPU Use

bdkjones, 2022-09-03:
**Describe the bug**

Al Dente constantly uses about 0.5-0.8 CPU, which is wasteful.

**To Reproduce**
1. Run Al Dente.

**Expected behavior**
Al Dente should not consume CPU cycles needlessly.

**Screenshots**
[image]

**MacBook:**
 - MacBook Pro (16-inch, 2021)
 - macOS 12.5.1
 - AlDente Free
 - AlDente 1.18
 - Default settings enabled; charge level set to 69% and maintaining.

**Additional context**
This looks like the sort of wasteful CPU usage you get with polling. I haven't looked through the codebase extensively, but I'd wager there's polling going on.  A constant use of 1% CPU for an app that simply maintains a certain battery charge level seems wasteful.

I'm also curious what the `AlDente Networking` process is and why there's a constant process dedicated to the apphousekitchen website. I understand checking for updates, but those processes should be closed and cleaned up once the update check completes. I don't understand why these two processes must exist at all times.

In short, the app seems to be a bit careless with system resources.

bdkjones, 2022-09-07:
Great; thanks for the info!

Is it possible to increase the polling interval? It would be nice to save all the CPU cycles we can.

MatthiasKerbl, 2022-10-04:
> 

Hi @bdkjones,

It would theoretically be possible to increase the polling interval. However, AlDente would become less responsive. Therefore, we had to find a sweet spot between responsiveness and CPU usage.

Cheers,
Matthias

MatthiasKerbl, 2022-11-01:
Hi @bdkjones,

I am converting this issue into a discussion since it is more of a general discussion than an actual issue.

Hope that's understandable.

Cheers,
Matthias