AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #768: ALdente seems to stop working after an update

donaldawalt, 2022-09-13:
**Describe the bug**
Yesterday, I updated 2 MacBook Pro's, one Intel- one Apple-based silicon, to version 12.6 of MacOS. I noticed today, again, instead of being around 60%/my setting, both laptops were pegged at 100%. This seems to have after MacOS updates.

Things I did to try and get it working again:
1) Click discharge - turns briefly orange to discharge, but very quickly pops back to blue.
2) Reinstall the Helper Application
3) Close and reopen Aldente

So now I just unplugged the laptops from power, and I'll wait until they get below 60%. I seem to remember that works.

**To Reproduce**
Steps to reproduce the behavior:
1. Perform MacOS update
2. Monitor Aldente - laptop will charge until battery gets to 100%

**Expected behavior**
Aldente should settle in to 60% battery charge, my setting.

**MacBook (please complete the following information): 
 - MacBook models - MacBook Pro 16" 2019 Intel, MacBook Pro 16" 2021 Apple M1 Max
 - macOS version 12.6
 - AlDente Pro or AlDente Free? AlDente Pro
 - AlDente Version? 1.18
 - List of all enabled/disabled settings in AlDente:
 -Enabled: Sailing mode, Hardware Battery Percentage, Updates Automatically, Launch at Login, % Percentage Icon, Show on Startup, Popup Animation
-Disable - everything else

**Additional context**
N/A

lsusanka, 2022-09-14:
I just updated my late-2013 15" MacBook Pro 11,3 to Monterey 12.6 from 12.5.1 (using OpenCore-Patcher.app to re-enable the graphics card) and now the free version of Aldente no longer works. I tried the usual things---reinstalling, removing helper app, deleting and reinstalling the application itself, reset settings etc. No joy. I was thinking of buying this app (it is quite cool) but ....

lsusanka, 2022-09-14:
By the way, when I say "no longer works" I mean that the charge light stays green all the time, at 100% even though Aldente was set to 80% (changed now to 50%.)

mouthzipper, 2022-09-15:
I have the same issue. M1 Pro. it keeps charging til 100% even though I set it to an 80% charge limit.

svoop, 2022-09-15:
Same here on another M1 Pro. Funny enough, my girlfriend has the same model (only with a bigger SSD) and AlDente appears to work nomally even after the update – at least it doesn't charge beyond the configured 80%. It's possible however that her computer was slightly below 80% at the time of the update and mine slightly above.

I'm trying to discharge below 80% by disconnecting the charger and then see if the recharge stops at 80% or goes all the way to 100% again.

Also I can confirm that reinstalling the helper doesn't fix this issue.

svoop, 2022-09-15:
@donaldawalt You might want to update the title of this issue to something like "AlDente stops working after update to macOS 12.6" to emphasize this obvious relation to the macOS update in the issues list.

donaldawalt, 2022-09-15:
@svoop done, I changed the title per your suggestion. Thanks!

svoop, 2022-09-15:
Update – I've tried the following: Using the battery (power disconnected) below the threshold (80%) and then reconnecting the power, AlDente does not stop the charging at the threshold. The behavior is exactly as described by the OP. 

I've got an AlDente Pro license, so I'll try to submit this issue to Pro Support as well.

lsusanka, 2022-09-15:
Update: I noticed that the finder toolbar item lost the discharge and top up buttons. I went to Activity Monitor and force-quit the AlDente process and restarted AlDente and they re-appered as usual.
[Screen Shot 2022-09-15 at 4.17.25 PM.pdf](https://github.com/davidwernhart/AlDente/files/9579536/Screen.Shot.2022-09-15.at.4.17.25.PM.pdf)
I have now disconnected power and ran the battery down to 9% and reconnected. I'll post again when I find out if it stops at 80%.

lsusanka, 2022-09-16:
It stopped at 80%     !!!!
Hopefully, my comment will give a hint for the software creator.

svoop, 2022-09-16:
@lsusanka Hmm, your fix hints at battery miscalibration. This would also explain why it happens on one Macbook but not another. Maybe the macOS has done something with the SMC. In this case, recalibration (easiest using the corresponding feature in AlDente) might help. I can't test this theory right now but will do as soon as possible.

donaldawalt, 2022-09-16:
Good info. To test your theory @svoop I had AlDente uninstalled on the one offending MacBook Pro (which I had just calibrated last week, but I think before the update to 12.6). I just installed it, activated my Pro license, and now I am doing a calibration. It was at 99% battery before I did this. I'll report back once calibration is done and I can determine if it's working or not.

donaldawalt, 2022-09-16:
Well the calibration did not work. What a mess AlDente has become for me. After 8 hours, it's still hanging on 99% charge, plug is green, and the progress bar is green not yellow even though it should be discharging. I have been using the laptop (plugged in) throughout the day.

So on a lark, I stopped calibration, closed AlDente, reopened it, turned off Hardware Battery percentage, closed it, reopened it, and started Calibration again. We'll see if that works, if not I may just uninstall it completely until this is fixed.

donaldawalt, 2022-09-16:
I noticed the process is not running when I just close the app from the menu bar, so I doubt that's it - as I have been closing it a lot. Calibration is not working, it keeps going to charging - even though the calibration page showed it on the "Discharge to 15%". Battery stays at 99%, the progress bar is green when I click on the icon on the menu bar, and the icon on the menu bar shows the little '+' sign not the '' sign. I used AppCleaner to uninstall everything too, prior. This is just not working right at all on an Apple M1 Mac, at least for me.

lsusanka, 2022-09-16:
When it was not working but running my menu bar showed no "Top up plus" or "Discharge minus" at all, just the slider bar, charge limit percent box and general settings button. But ... it has now been operating for me for a day ....

donaldawalt, 2022-09-16:
I am going to try one more thing, then I give up. AlDente is again completely uninstalled using AppCleaner. I am going to run on battery now until it gets down to 10% - 20%, then I will install AlDente and see if it stops at 60% (where I set it) or blows past it to 100%. I should know by tomorrow, I'll post back!

donaldawalt, 2022-09-18:
I am in the second day since I plugged back in, it got to 15%, then I ran AlDente, and plugged in the MagSafe adapter to start charging. I had it set to 60% with Sail Mode 10%, and it stopped at 15% this time! So this area needs work, maybe it's a one-off and things will sail smoothly again, Thanks for everyone's help, I hope this report helps the AlDente authors to narrow the problem down!

spidey-mx, 2022-09-19:
Seems that I got a temp fix. I reinstalled the app (through Aldente app) and I restarted the laptop. Somehow the percentage jumped from 96% to 100%, but after that when I clicked on Discharge button, Aldente works again. I have been working around 30 minutes and not charging message is not coming again. I will keep you posted for any update.

svoop, 2022-09-19:
Here's what the friendly guys from AlDente support wrote in reply to my question:

> macOS updates can sometimes change some weird SMC variables, which makes AlDente temporarily ineffective. We are investigating what the update process actually changes and how to prevent this issue, but unfortunately, we have not found the culprit yet.
>
>This being said, the fix usually simply is to power off the MacBook and close the lid for 2 minutes. This should reset the SMC on your device. Also, please unplug your power adapter from the wall and from the dock and wait 30 minutes before plugging it in again. This has resolved similar issues in many cases before. 

I'll try this later today, but wanted to post it here for others to give it a shot as well.

donaldawalt, 2022-09-19:
@svoop Thanks for sharing! Mine has been fine now for two days after I went through some convoluted steps which I outlined above. I have kept this response for future reference if/when it happens again!

Good luck getting your working!

MatthiasKerbl, 2022-09-20:
Hi everyone,

We are sorry that so many of you had issues with AlDente after the newest macOS update.
We are aware of this issue but did not find a solution yet. However, we will keep working on it and give updates here.
I have also updated our FAQs about this topic:

_**AlDente stops working after the macOS update!**

Unfortunately, AlDente temporarily does not work after a macOS update for a small number of users.

To get AlDente to work again, please try the following steps which have worked for many users before:

1. Unplug and discharge your MacBook to less than your current charge limit in AlDente
2. Turn off your MacBook and close the lid for a couple of minutes
3. Unplug the power brick from the outlet and wait a couple of minutes
4. Turn on your MacBook and plug it in again
5. Check if it stops charging at the charge limit set in AlDente

We are aware of this bug for a while now but were so far not able to fix it. However, we will keep trying._

One possibility to actively avoid this issue during the next macOS update is to close AlDente completely before the update and restart it when the update is finished.

Cheers,
Matthias

donaldawalt, 2022-09-20:
Thank you @MatthiasKerbl for the detailed reply!

clementranc, 2022-09-22:
Hello, 
Following @MatthiasKerbl suggestion, I have been able to get AlDente Pro 1.18 (on macOS 12.6) to stop the charge at the limit. However, nor the discharge or the sailing mode works. I mean that the best behavior I get is: once the charge limit is reached, the charge is paused.

cagnulein, 2022-09-30:
unfortunately same here :( is there any update about this?

CometYZ, 2022-10-01:
Me too, but only for Ventura beta 9. It works fine in 12.6 (macbook air m1)

andrewcrook, 2022-11-01:
working fine for me M1 MBP 13”, 8 cores ,16GB, 1TB, however, I do power off at night.

alaneuler, 2022-11-05:
Is it related to the change of SMCkeys? https://github.com/beltex/SMCKit/issues/48

MatthiasKerbl, 2022-11-05:
> Is it related to the change of SMCkeys? [beltex/SMCKit#48](https://github.com/beltex/SMCKit/issues/48)

Hi @alaneuler,

Thanks for the hint but I doubt that it is connected with this.

Our best guess is that macOS is going into some kind of safety state during the macOS update process which prohibits AlDente from influencing the charging behavior. For some reason, this safety state seems to persist for some users even when the update is finished. However, following the steps mentioned above seems to disable this safety state of macOS.
We are currently investigating this issue and trying to come up with a solution.

Cheers,
Matthias

mimkorn, 2022-12-15:
> Hi everyone,
>
> We are sorry that so many of you had issues with AlDente after the newest macOS update. We are aware of this issue but did not find a solution yet. However, we will keep working on it [...]

Hi, Matthias,

I find the steps a bit confusing in one part:

1. One is to discharge MacBook below charge limit (which with this bug means disconnecting from power outlet).
2. Then you're to turn off and close the lid 
3. and then to disconnect the power outlet. 
 
So should I connect the outlet before turning off and closing the lid? Or is the step with disconnecting power outlet redundant cause it's already disconnected?

Also, wouldn't a full reinstall of AlDente fix it as well?

Thanks, 
Šimon

MatthiasKerbl, 2022-12-24:
Hi @mimkorn,

> So should I connect the outlet before turning off and closing the lid? Or is the step with disconnecting power outlet redundant cause it's already disconnected?

It is important to unplug the MacBook from the power brick and unplug the power brick from the socket. In which order you are doing these two steps does not matter.

> Also, can one set the limit up to an arbitrary high percentage to speed up going through this workaround? E.g. to 97%

Yes.

> Also, wouldn't a full reinstall of AlDente fix it as well?

No, because the issue lies within macOS and not AlDente. Therefore, reinstalling AlDente does not solve the issue.

Cheers,
Matthias