AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #773: Strange and useless slider in the status bar menu

Volodymyr-13, 2022-09-18:
**Describe the bug**
I just found this slider for % is very awkward and useless. Just remove it.

**To Reproduce**
It's doesn't work, this bug is forever there. 
Video attached.

**Expected behavior**
Remove that awkward slider, which is useless. 

**Screenshots**

https://user-images.githubusercontent.com/3391456/190918239-7dd0f678-7c49-4188-b0cf-b590698b185a.mov

https://user-images.githubusercontent.com/3391456/190918242-748dae2f-3a9c-4ef0-963a-55aa492b3eae.mov

**MacBook (please complete the following information):**
 - MacBook 14" M1 2021
 - macOS version macOS 12.6
 - AlDente Pro 
 - AlDente Version 1.18
 - List of all enabled/disabled settings in AlDente:

[image]

MatthiasKerbl, 2022-09-20:
Hi @Volodymyr-13,

First of all, the slider is one of the main and most essential features of AlDente since it allows our users to control the Charge Limit feature.
You have two options to control the charge limit. Firstly, you can do that with the percentage box by writing the desired charge limit and clicking enter. Secondly, you can just use the slider to change the charge limit.

Regarding your first video, while writing in the percentage box, you can not use the slider. You need to click enter first after changing the charge limit for the slider to work again.
Regarding your second video, it seems like something is off with AlDente on your MacBook. I would recommend completely reinstalling it again. (https://apphousekitchen.com/how-to-uninstall-aldente-pro/)

Hope that helps.

Cheers,
Matthias

Volodymyr-13, 2022-09-20:
> First of all, the slider is one of the main and most essential features of AlDente since it allows our users to control the Charge Limit feature.
This is strange, since once you set 80% you can/should forget about any limits forever, so because you're using sailing mode to preserving battery life, you don't need to change it ever. This slider can be replaced maybe with 2 buttons/or one switch like 80%(configurable) and 100%, that's it..
My MacBook is always on charge, for last 6 months of using AlDente I never touched that slider, just recently accidentally clicked on it and found that glitches.. 
I can't imagine a real use case situation where it's required from user to set 85%, 94% or 32% or anything else besides 80% and 100%... 

That slider was always glitchy, reinstalling doesn't helps. I already reported some bugs with it, but looks like this can't be easily fixed, since having focused active input field overrides slider and clicking/activating/changing slider element doesn't focuses only it, but stay input field focused..

MatthiasKerbl, 2022-09-20:
> This is strange, since once you set 80% you can/should forget about any limits forever, so because you're using sailing mode to preserving battery life, you don't need to change it ever. This slider [...]

Hi @Volodymyr-13,

Thank you for your input.
Many users including me like to use the slider. If you do not need it, you can just ignore it. Another feature of the slider is to show the current charge level in relation to the charge limit and also the current charging state.
The only thing that is not 100% coherent in regards to the slider for me is messing with the slider while the percentage box is in focus. We will try to fix this. However, it is not a priority right now since it is an edge-case scenario because usually one would either use the slider or the percentage box and click enter afterward and not both at the same time.

Cheers,
Matthias

MatthiasKerbl, 2022-10-24:
Hi @Volodymyr-13,

We have just released AlDente 1.19.1 which will now properly unfocus the percentage box when the slider is moved while the percentage box is in focus.

Cheers,
Matthias

closed 2022-10-24 by MatthiasKerbl