AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #780: macOS's battery optimisation no longer works after uninstalling AlDente

daryll-swer, 2022-09-23:
**Describe the bug**
After uninstall AlDente helper and the app itself, macOS battery optimisation does not work even after 2 weeks straight and yes I have enabled/disabled it and also drained battery to 0% and charge it back up to 100% three times in a row.

Did AlDente broke something at system level here?

**To Reproduce**
Steps to reproduce the behavior:
1. Install AlDente
2. Remove AlDente

**Expected behavior**
macOS default behaviour for battery optimisation

**MacBook (please complete the following information):**
 - M1 Air
 - macOS version 12.6
 - AlDente Free?
 - AlDente Version - 1.18
 - List of all enabled/disabled settings in AlDente: Defaults

MatthiasKerbl, 2022-09-23:
Hi @daryll-swer,

> Did AlDente broke something at system level here?

It is highly unlikely that AlDente "broke" something on the system level. If you uninstall AlDente completely and perform an SMC reset (which is done automatically on Apple Silicon Macs when you shut them down and close the lid for more than 30 seconds), all traces of AlDente are removed from your system.
My guess is that you are one of the many users for whom the macOS Battery Optimisation feature does not work as intended or it just needs more time than two weeks to learn your schedule.

Cheers,
Matthias

daryll-swer, 2022-09-23:
@MatthiasKerbl 

The macOS battery optimisation feature was working perfectly fine for my device before installing AlDente, and it only took a few days (not weeks) to become active. And yes, I've done the “SMC reset” with both power plugged in and unplugged after shutting the device down.

It does seem that something went wrong after installing AlDente.

daryll-swer, 2022-09-30:
@MatthiasKerbl Do you have any recommendations other than reinstalling macOS from scratch (which would be tedious) to fix this? It's now been 4 weeks and battery optimisation on macOS is still not working like it used to in the past before AlDente was installed.

daryll-swer, 2022-10-09:
Since there was no update on this, I had to do a macOS reinstall by wiping the disk. Battery optimisation started working again in just 6 days.

closed 2022-10-09 by daryll-swer