AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #782: Aldente Pro Won't Discharge Below 20% M2 MBP 13

nbokh, 2022-09-29:
**Describe the bug**
Aldente Pro doesn't discharge below 20%

**To Reproduce**
Steps to reproduce the behavior:
Discharge below 20%

**Expected behavior**
MacBook should discharge below 20%

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model 2022 M2 MBP 13 [base model]
 - macOS version (https://support.apple.com/en-us/HT201260)
 - AlDente Pro or AlDente Free? Pro
 - AlDente Version? 1.18
 - List of all enabled/disabled settings in AlDente:
Everything enabled except [sailing and heat protection]
**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2022-09-29:
Hi @nbokh,

This is actually by the design of the Discharge feature of AlDente.
The reasons why we chose 20% as the lower limit are the following:

1. Some studies show that keeping your battery under 20% will result in faster battery degradation.
2. MacBooks with very old batteries or miscalibrated batteries often shut down between 10-20%.
3. Even though, there are more safety measures in place, we do not want people to use the Discharge feature to discharge their MacBook completely to avoid any weird bugs.

What would be the reason you want to see the Discharge feature to work below 20% and how far down would you like it to work?

Cheers,
Matthias

nbokh, 2022-09-30:
Hi @MatthiasKerbl the reason I had tried to go below 20% was because of an unsuccessful `Battery Calibration`. During the day I use an external monitor so it really takes all day to go from 100% ➡️ 15%... Now it's nighttime. So naturally the next day I'd like to continue the calibration, but it seems the Aldente Pro app doesn't allow me to continue the calibration. So me wanting to go below 20%, I am trying to do the calibration myself [if that is even possible]

MatthiasKerbl, 2022-09-30:
Hi @nbokh,

You can always do a manual calibration. Therefore, you need to do a full battery cycle manually 0%-100%-0%.
The lower limit during the Calibration mode of 15% is a chosen middle ground for all the variables that played a role during the development.

Cheers,
Matthias

closed 2022-09-30 by MatthiasKerbl

nbokh, 2022-09-30:
@MatthiasKerbl yeah but I can't actually do a manual calibration with the Aldente app because of this limit [20%]. I also can't do it with my current setup as my thunderbolt hub also provides power to the MacBook.

MatthiasKerbl, 2022-09-30:
> @MatthiasKerbl yeah but I can't actually do a manual calibration with the Aldente app because of this limit [20%]. I also can't do it with my current setup as my thunderbolt hub also provides power to the MacBook.

In this case, you can try to restart the calibration mode of AlDente or use your MacBook completely unplugged to do a manual calibration.

Cheers,
Matthias