AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #783: Hide menu icon being a pro feature is not fair.

parsanoori, 2022-10-05:
AlDente is actually the only working charge limiter for me.
I use the 2021 macbooks which have notch bar. Hence AlDente's icon being in the menu bar is a pain.
I live in Iran. I cannot pay for the pro version since neither financially affordable nor even possible to pay. That's because Iran has a  failed economy and it's under heavy US sanctions, therefore no online payment method is available.
AlDente' hide menu icon feature being a pro feature is not fair for so many people like me.

DaJiaHaoWoShiSZ, 2022-10-17:
With all due respect, you can afford a MacBook Pro starting at $1999 while you can't afford a life-time license with $29?

MikixIT, 2022-10-19:
@DaJiaHaoWoShiSZ 
> With all due respect, you can afford a MacBook Pro starting at $1999 while you can't afford a life-time license with $29?

with all due respect, so i buy a mac at 1k do i have to spend for everything? maybe I bought the pc saving all my savings? For something that should be open source?

parsanoori, 2022-10-28:
@DaJiaHaoWoShiSZ 
Third parties and smugglers can buy physical products and import them to the country. So I can buy a macbook.
But since there's no store to do global payment only for me, I can't buy softwares.
Laptop is a must for many people. One should pay for it. But not is every app is as must have as a laptop itself to go buy it.

a-plastic-bag, 2022-11-08:
You could always use [Dozer](https://github.com/Mortennn/Dozer).

MatthiasKerbl, 2022-11-11:
Hi @parsanoori,

Since the release of AlDente Pro, most improvements and new features are only available to AlDente Pro users. On the other hand, we are still improving and maintaining the free version as well. This does not mean that we might not change some of the available settings/features in the long term. When it comes to the possibility of disabling the menubar icon, we are having no plans currently on offering that in the free version as well. Furthermore, there are many options to deal with them and some are even free:
- If the AlDente menu bar icon takes up too much space, just drag and drop it to the left side and it will probably be hidden behind the notch.
- Disable other macOS menu bar icons you do not need in macOS settings.
- Use third-party apps such as Dozer or Bartender to manage your menu bar.

Hope that helps.

Cheers,
Matthias

closed 2022-11-11 by MatthiasKerbl

MikixIT, 2022-11-11:
So @MatthiasKerbl , the solution is to download additional third party apps to fix a problem generated by your own app? Wow, I've never heard this one before. 
You know what I tell you, better not to use your app, it seems that you just want to make money on it.
And to clarify, earning on a developed app, that's fine. But putting such a feature like that for a fee is just a joke.