AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #784: while the al dente is in calibration mode, turning off the monitor does not work.

Namoki, 2022-10-10:
I'm using an M2 MacBook Air. I am using Aldente Pro version 1.18.

However, while the al dente is in calibration mode, turning off the monitor does not work.

When charging and discharging in calibration mode, the monitor off time set in the MacOS control panel does not work at all.

For a long time during the calibration mode, the external monitor remains on all the time.

Please test it.

MatthiasKerbl, 2022-11-05:
Hi @Namoki,

Unfortunately, AlDente needs to disable sleep during the Calibration Mode. This is due to third-party apps not being allowed to run while macOS is in sleep. While sleep is disabled, the macOS setting Lock Screen is disabled as well.
We will try to get Lock Screen to work again even if sleep is disabled. However, we are not sure if this is possible at all.
In regards to disabling sleep during Calibration Mode, we can not do anything against that since it is a restriction coming from macOS.

Cheers,
Matthias

MatthiasKerbl, 2025-02-21:
Hi @Namoki,

We have just released AlDente 1.31, which will now consider the Lock Screen and Screensaver settings from macOS.

Best,
Matthias

closed 2025-02-21 by MatthiasKerbl