AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #785: Add a `Get Charge Limit` shortcuts action

infused-kim, 2022-10-11:
**Is your feature request related to a problem? Please describe.**
I would like to build a [raycast](https://www.raycast.com) extension that shows the currently set charge limit.

**Describe the solution you'd like**
Add a `Get Charge Limit` shortcuts action that returns the current charge limit as a percentage value.

**Describe alternatives you've considered**
I tried all the existing actions, but none of them do what I want. The `Get Percentage` action returns the current charge level and now the limit.

**Additional context**
[image]

MatthiasKerbl, 2022-10-14:
Hi @infused-kim,

That's a good idea and we will implement it with one of the future updates.

Cheers,
Matthias

MatthiasKerbl, 2022-10-16:
Hi @infused-kim,

We have just released AlDente 1.19 which includes Get Charge Limit shortcuts action.
Hope you like it.

Cheers,
Matthias

closed 2022-10-16 by MatthiasKerbl

infused-kim, 2022-10-18:
Amazing. It's working perfectly.

Btw, the blog post that is linked in the update note about automatic discharging doesn't seem to exist.

MatthiasKerbl, 2022-10-18:
I am glad you like it.
That's weird, it is working for me here: https://github.com/davidwernhart/AlDente/releases