AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #786: Still stop charging with "Stop charging when sleeping" off

Rrobinvip, 2022-10-14:
**Describe the bug**
As the title suggests, ALDente still stops laptop from charging even when "Stop charging when sleeping" is off (I have basic version installed so it's off whenever). 

**To Reproduce**
Steps to reproduce the behavior:
The way I put my laptop to sleep:
I have an external display, so I always manually click "sleep" at the top left menu by clicking apple icon. After the laptop goes sleep, I unplug my external display and leave my laptop for charging. 

**Expected behavior**
Battery charged will ignore restrictions when sleeping.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model: MacBook Pro 15' 2018
 - macOS version: macOS 12.6
 - AlDente Pro or AlDente Free: free
 - AlDente Version: 1.18
 - List of all enabled/disabled settings in AlDente: General settings (check automatically - on, launched at login - on), charge setting (all off), appearance (show on startup - on)

**Additional context**
This issue doesn't occur 100%. But happens occasionally.

MatthiasKerbl, 2022-10-21:
Hi @Rrobinvip,

Let me explain how AlDente Free is behaving in regard to sleep:

- When your set charge limit in AlDente is reached and charging is paused and you activate sleep, your MacBook will not start charging while in sleep. Instead, it will keep charging paused.
- When your set charge limit in AlDente is higher than your current charge level of your MacBook and your MacBook is currently charging when you activate sleep, your MacBook will continue charging while it is in sleep and even overshoot the set charge limit in AlDente and will continue charging to 100%.

Do these explanations explain the behavior of AlDente Free in your case?

Cheers,
Matthias

MatthiasKerbl, 2022-11-01:
Hi @Rrobinvip,

Any update on this?

Rrobinvip, 2022-11-01:
Thank you for explanation!

closed 2022-11-01 by Rrobinvip