AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #788: Did no receive the license on email

theo-raducanu, 2022-10-16:
@davidwernhart @MatthiasKerbl I purchased one day ago aldente pro and I did not receive the license on email after 15 minutes, i wrote to support and did not receive any message. Can you please help with that?

MatthiasKerbl, 2022-10-16:
Hi @theo-raducanu,

I am sorry you are having trouble with the buying process.
Please take a look at the following steps from our FAQ:

**I Did Not Get The License Code Or Any Receipt Emails After Buying AlDente Pro!**

There are multiple reasons why you did not get your license code after the purchase:

1. Sometimes it takes up to 30 Minutes for the purchase emails to go through.
2. The purchase emails end up in the junk email folder.
3. You had a typo in the email address during your purchase.
4. Failures of the payment networks/providers.

Please contact the support team of our payment provider Paddle at [assist@paddle.com](mailto:assist@paddle.com) and include the email address you used to purchase AlDente Pro. The Paddle support will try to solve your issue as fast as possible.

Cheers,
Matthias

Volodymyr-13, 2022-10-19:
It's in SPAM folder, same was for me. 

I already created some bug/feedback about paddle and it service/support, don't remember where..
Afterwards, I'll never ever buy anything via Paddle... so far I have 3 apps in my list, which I can't buy due to its using paddle as payment method.

MatthiasKerbl, 2022-11-06:
Hi @theo-raducanu,

Do you still need help with this or were you able to resolve it?

Cheers,
Matthias

MatthiasKerbl, 2022-11-11:
Hi @theo-raducanu,

Since I haven't heard back from you in a while, I will close this issue for now.
I Hope you could resolve this issue with support from Paddle.
Let me know if you still need help.

Cheers,
Matthias

closed 2022-11-11 by MatthiasKerbl