AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #794: Automatic Discharge and Discharge are not working at all

Volodymyr-13, 2022-10-19:
**Describe the bug**

https://user-images.githubusercontent.com/3391456/196671958-6829a5f6-d65b-497f-a20f-142a5d17ea08.mov

**To Reproduce**
Steps to reproduce the behavior:
1. just try to enable Automatic Discharge or Discharge

**Expected behavior**
Should just work.

**MacBook (please complete the following information):**
 - MacBook Pro 2021 M1
 - macOS version 13.0 (22A379)
 - AlDente Pro
 - AlDente Version 1.19
 - List of all enabled/disabled settings in AlDente:

[image]

Volodymyr-13, 2022-10-19:
Was unable to attach video(too high size) while I enabled Automatic Discharge option and nothing happens. 
Either way, it just doesn't work.

Volodymyr-13, 2022-10-19:
After a couple of app restarts it's now working, but due to [another bug](https://github.com/davidwernhart/AlDente/issues/793) I can't use this option.

MatthiasKerbl, 2022-10-19:
> @MatthiasKerbl why you always so fast inclosing tickets without conformation.... what the point?
> 
> It's set to 80% and current charge is 84.24%
> 
> Don't u see this on the video?
> 
> [image]

Hi @Volodymyr-13,

Sorry if you felt wronged, in my opinion, the issue was solved.

The battery percentage used in AlDente is 80% which is indicated by the blue bar and you can also activate it to show the percentage in the menubar in the Appearance settings in AlDente.
The reason, why you are seeing a different battery percentage in the menubar is most likely because you are using a different app than AlDente to show the battery percentage and this other app is most likely using the macOS battery percentage instead of the hardware battery percentage which you have most likely activated in the settings of AlDente. Please read more about this topic on our blog here: https://apphousekitchen.com/feature-explanation-hardware-battery-percentage/

Cheers,
Matthias

Volodymyr-13, 2022-10-19:
@MatthiasKerbl 
I'm using Wattagio because it's showing % with just more digits, so like 84.15%, your app shows only 84%, but it's the same in real value. 
[image]

So, at the moment when this bug occurs there was 84.24%, Al Dente shows 84%.

Also, I really waiting for the such precise info like from Wattagio https://github.com/davidwernhart/AlDente/issues/658

DamonHD, 2022-10-19:
Such a precise value **does not exist**.  It is a fiction.  Probably even the second significant figure shown is wrong.  These are all estimates by different bits of software with their own heuristics.

Adding more digits to the displayed value does not add more information, just noise.

Rgds

Damon

MatthiasKerbl, 2022-10-19:
> 

@Volodymyr-13 

We are only showing full percentages because it is accurate enough for now in our opinion.

Did you try to use Discharge now as intended and did it work for you?

More information about the battery and charging details is definitely something that makes sense. However, it is not a priority right now.

closed 2022-10-19 by MatthiasKerbl

Volodymyr-13, 2022-10-19:
> Such a precise value **does not exist**. It is a fiction. Probably even the second significant figure shown is wrong. These are all estimates by different bits of software with their own heuristics.
> 
> Adding more digits to the displayed value does not add more information, just noise.
> 
> Rgds
> 
> Damon

So why Wattagio and coconutBattery showing these values? 

[image]

I see that coconut is rounding a bit, but anyway they just both taking this value from macOS raw output via some api..

Volodymyr-13, 2022-10-19:
> Did you try to use Discharge now as intended and did it work for you?

https://github.com/davidwernhart/AlDente/issues/794#issuecomment-1283842291

I just restarted app 2-3 times.

DamonHD, 2022-10-19:
People will paint go-faster stripes on your car if you ask, but they won’t make your car go faster.

Many people do not understand (a) actual information content of numbers in general, and (b) how hard and rough capacity estimation of in-use batteries actually is.

I’ve spent about a decade tuning parameters for one boring old lead-acid battery bank, and indeed the previous battery set died before I was happy with the accuracy…  I’m starting again with the new set.

Rgds

Damon

Volodymyr-13, 2022-10-19:
Oh man @MatthiasKerbl again u closed this ticket? 
Suddenly this is not a bug? 
You're not really worried that I didn't changed anything besides 2-3 times just restarted your app? 
Typical behaviour for Pro version? Just a couple app restart and maybe it will work like it should?  
I updated it from ~1.18 to this 1.19 and got that bug.
 
I see quality of this app becomes low and you less care about bugs.

DamonHD, 2022-10-19:
You are rude and wrong.

MatthiasKerbl, 2022-10-20:
> Oh man @MatthiasKerbl again u closed this ticket?
Again, sorry if you feel wronged. This is not my intention. However, you reported in my understanding that everything was working.

> Suddenly this is not a bug?
Could you give me a precise description of the bug and how to reproduce it? Because as far as I understood it, everything you reported above was never a bug but a misunderstanding about how AlDente Pro works.

Volodymyr-13, 2022-10-20:
When battery was 84% and charge limit was set to 80(it always that, and I never ever changing it) - any discharge function  didn't worked. I already showed this on the video and tried to explain in some posts before. 
-----
Can you remove that spam/offtopic posts from here from some random person about cars and so on?
Since we discussing a bug here.

MatthiasKerbl, 2022-10-20:
> ## When battery was 84% and charge limit was set to 80(it always that, and I never ever changing it) - any discharge function didn't worked. I already showed this on the video and tried to explain in some posts before.

I am not seeing a bug in the video you sent and I already explained what was going on in this situation in the comments above but I can paste it here again:
> Regarding your issues about Discharge and the video attached: The reason why it does not work in your case is that your charge limit is set at the current battery level. If AlDente would activate Discharge in this scenario, it would not know when to stop Discharge again. In order to use Discharge properly, you have to set the Charge Limit lower than the current battery level and activate Discharge afterward. Then, AlDente will Discharge your MacBook to the set Charge Limit. Please read the feature description about Discharge on our blog: https://apphousekitchen.com/feature-explanation-discharge/

> The battery percentage used by AlDente in your video is 80% which is indicated by the blue bar and you can also activate it to show the percentage in the menubar in the Appearance settings in AlDente.
The reason, why you are seeing a different battery percentage in the menubar is most likely because you are using a different app than AlDente to show the battery percentage and this other app is most likely using the macOS battery percentage instead of the hardware battery percentage which you have most likely activated in the settings of AlDente. Please read more about this topic on our blog here: https://apphousekitchen.com/feature-explanation-hardware-battery-percentage/

Do you understand my explanation or do I need to explain it again?

Volodymyr-13, 2022-10-20:
I'm not using different battery percentage in any such apps, all them shows raw data only.

Wattagio and AlDente are configured the same and outputs the same value.
Just in your app this option called "Hardware Battery Percentage"(which I requested, by the way) and it's enabled and never been disabled. 
And in Wattagio it's called "Read out the raw battery data" which always enabled.

[image]

Do you understand my explanation or do I need to explain it again?

MatthiasKerbl, 2022-10-20:
> I'm not using different battery percentage in any such apps, all them shows raw data only.
> 
> Wattagio and AlDente are configured the same and outputs the same value. Just in your app this option called "Hardware Battery Percentage"(which I requested, by the way) and it's enabled and never been disabled. And in Wattagio it's called "Read out the raw battery data" which always enabled.
> 
> [image]
> 
> If I didn't disable % in your app, then it would show 84% on the video from initial post.
> Do you understand my explanation or do I need to explain it again?

@Volodymyr-13
AlDente does not use data from other apps. Therefore, I can only speculate on why other apps show different values than AlDente. The fact is, that AlDente is only using data directly from the MacBook, and this is consistent with the intended functionality of the Discharge feature in your video. Therefore, I can not see a reproducible bug of AlDente, and hence, this is also why I closed this issue.
If you find an inconsistency within AlDente or the intended functionality within its features, let me know.

Volodymyr-13, 2022-10-20:
Oh my, this is ridiculous. Don't u understand that all these 3 apps gets same data from MacBook on the same MacBook? 
Is that hard to understand from the screenshoot? 
I have no idea why your app was glitches at that moment, but this is bug anyway and it's occurred right after the update.

Your tone of communication certainly shows that you are a layman, this is not how you communicate with customers.
You should try to find out the real reason why the client contacted you and satisfy his needs.

You should send me a typical survey about this ticket and I'll fill it: "Bad, I'm not satisfied" and rate 0/5 stars.