AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #796: Wasn't able to purchase a Pro license

romanpost, 2022-10-21:
Hi!
@davidwernhart  thank you for a very useful app, indeed.
But unfortunately, I can't pay you for a great job you are doing. Let me describe why.

Yesterday I tried to pay for the pro License with my regular VISA card. 
As usual I switched on a Email relay on iCloud to generate a unique email. And use it for AlDente only.
And, to my surprise, my VISA was charged with following email notification:

"Your order for AlDente Pro is in process
You recently placed an order for AlDente Pro from AppHouseKitchen via Paddle.
During the order process, our automated system noticed a few characteristics of the order that mean that we need to manually review it before we can process the order further.
This process typically takes a few hours but will often happen much more quickly. No further action is required on your part. Don't worry, we haven't yet charged your card for the order. Once we complete our review, we'll process your transaction.
We operate this process to protect awesome customers like you from potential fraud."

OK, fraud is bad, no worries. Paddle obviously doesn't  like email privacy, so I don't care. But why there's a statement that my card is not charged when it's actually charged??

And after 4 hours another email:
"Unable to process your order for AlDente Pro
You recently placed an order for AlDente Pro from AppHouseKitchen via Paddle.
Your order was placed for manual review and unfortunately we have been unable to process it on this occasion.
Don’t worry, we have not charged your card for this order and all funds are still available to you."
And they made a charge back to my VISA card.

And finally - today I tried to make a payment once again, now without iCloud email relay. Just main gmail account.
And paddle blocked my card, after yesterday issues, obviously. So I can't use it for payments throught them, including AlDente

Well, that's the end of story.

The real problem is that Paddle doesn't use 3DSecure for internet payments - when 2FA as SMS code is being sent, to verify card ownership. If you are in talk with them - pls ask them to switch that on for you, as a merchant. You are loosing sales and clients.

MatthiasKerbl, 2022-10-22:
Hi @romanpost,

I am glad you like our app and thank you for your support. Also, I am sorry that you had big issues with our payment provider Paddle. I have forwarded your message to Paddle and will let you know what they have to say.

Cheers,
Matthias

MatthiasKerbl, 2022-10-24:
Hi @romanpost,

In order for Paddle to be able to assess what went wrong during your payment process they need your email address.
Could you send me your email address? Then I will forward it to Paddle.
If you don't want to post your email address here, you can send it to me at news@apphousekitchen.com.

Cheers,
Matthias

MatthiasKerbl, 2022-11-06:
Hi @romanpost,

Any news on this?
Paddle is still waiting for your email address.

Cheers,
Matthias

MatthiasKerbl, 2022-11-24:
Hi @romanpost,

I will close this issue for now but feel free to report back if you have any updates.

Cheers,
Matthias

closed 2022-11-24 by MatthiasKerbl