AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #797: Lost ~900 mAh full capacity after using aldente

Delvien, 2022-10-22:
After using and uninstalling aldente, my macbook pro m1 16" lost around 1000 mAh of the full capacity. When it charges it charges to 7700~ mAh, but the full design is around 8600~
[image]

The battery health is still showing good, however.

Maybe I am missing a way to reset this after removing aldente?

I have started a calibration charge, but due to the m1's battery life, I have to leave it plugged in for almost 24 hours to complete a full discharge.

Delvien, 2022-10-24:
I've done two full recharges since uninstalling aldente, and the situation hasnt gotten any better:

[image]

Am I missing some magic sauce?

Delvien, 2022-10-24:
I've "reset the SMC" by rebooting once plugged in, even shut it down when I was plugged in. No change.

It doesnt seem plausible that I would drop ~900 mAh within the span of 2 days.  I started using aldente on this mac 10/19 or so.

[image]

My battery health cant magically "go up" so Im wondering if something that aldente did is lingering? I dont know what this does on the backend. :)

Volodymyr-13, 2022-10-24:
What is the charging limit you have set? 
What system settings shows for Maximum Capacity?
Even if it was 100%, such degradation is not normal, so I can assume that this is a defective battery.

Delvien, 2022-10-24:
80% 

The thing is. I keep a close eye on my battery full capacity, because I'm super anal about it apparently, which is what prompted me to use Aldente. Trying to preserve my battery for the long-haul. 

I was at 8542 mAh full capacity > Started using Aldente > 2 days later I was missing 900 mAh full capacity > here we are.  That's an unbelievable coincidence that I refuse to give any legitimacy to.

Volodymyr-13, 2022-10-24:
For this or any other app it's impossible to change capacity in such way.
You should ask Apple about your battery and warranty.

Delvien, 2022-10-24:
Did you develop Aldente? 

Do you know what it does to manage the charge limits?  Does it write to the SMC about what the capacities are to do this?

I'm not a programmer, but if I were a betting man, My bet is that it's not the battery. Problem > Cause > Effect

Volodymyr-13, 2022-10-24:
Nope.

gselsidi, 2022-10-29:
> I've "reset the SMC" by rebooting once plugged in, even shut it down when I was plugged in. No change.
>
> It doesnt seem plausible that I would drop ~900 mAh within the span of 2 days. I started using [...]

what? From your picture it shows you capacity went up in 2 days.....  from 10-21 to 10-23, your original capacity was going down steadily before you even started using Aldente..... assuming you started on 10-21 your capacity went up.....

really confused at what you're saying here.....

Delvien, 2022-10-29:
Meaning the actual.. physical... capacity, wont magically go up. It's calculation should, if the battery is ACTUALLY 98%..but system is reading it at 89%~. 

My guess would be whatever aldente does makes the system think it's full capacity is actually lower than it is.  But again, thats just a guess, im not a coder.

All I see is that using aldente made my system "forget" about almost 900 mAh capacity. Again I will reiterate, my capacity was sitting at 97% before I used it. 2-3 days later, it was at 87%. The only thing that changed was using this piece of software.

gselsidi, 2022-10-29:
Based on what? The screenshot you provided shows 04/22 nothing for 3 months… and shows 10/21…. At 7600

Are you saying right before 10/21 you were at 8600 capacity….

Delvien, 2022-10-29:
> Based on what? The screenshot you provided shows 04/22 nothing for 3 months… and shows 10/21…. At 7600
> 
> Are you saying right before 10/21 you were at 8600 capacity….

10/20 or 10/19 yes. The screenshot is from coconut battery. I dont normally use that, I use "battery monitor" which I look at every day, but it doesnt have a history. On those, or around those dates, I was around 8600 (probably 85** give or take) that I saw with my own two eyes.

Unless you have another way of seeing historical data on battery capacity on a system level? Coconut battery only logs it if the application is opened. Hence the dates being so far apart.

gselsidi, 2022-10-29:
Try this see if it helps.

https://www.iphonetricks.org/check-battery-health-on-mac-using-terminal/

Delvien, 2022-10-29:
> Try this see if it helps.
> 
> https://www.iphonetricks.org/check-battery-health-on-mac-using-terminal/

Right, and that says the same thing battery monitor did, Im looking for historical logging in the system level. 

This is only really to show you logging wise.. proof  that there was a drop in maximum capacity if you dont want to take my word for it.

gselsidi, 2022-10-29:
No idea wait for dev to respond

gselsidi, 2022-10-29:
FYI coconut is reporting wrong battery health, OSX reports 89% while Coconut is 81%, so right there shows it's not as reliable either way i wouldn't be to concerned as there is no way for software to physically lower your battery capacity. With no historical logging it's probably going to be very hard to pinpoint anything.

Bachsau, 2022-11-11:
Did you follow all the steps in the [uninstall guide](https://github.com/davidwernhart/AlDente-Charge-Limiter/issues/797) or have you just deleted the app?

Delvien, 2022-11-11:
> 

I did not. I just followed those steps after installing again, let's see if that fixes.

breytex, 2022-12-05:
@Delvien 
did it solve your problem?

Delvien, 2022-12-05:
> @Delvien
> did it solve your problem?

Sadly no. Apple confirmed thats how much total capacity I lost, couldn't tell me why but did say its not typical for a mbp this young. 

So yay me.

Bachsau, 2022-12-05:
If that's so, ask them to replace it. It is obviously defective.

closed 2022-12-05 by Delvien

Bachsau, 2022-12-05:
That might be because "everyone" else, like me, have been using AlDente for more than a year now without any such effect. But if you feel better if you can blame it on the app, go along with it.

Delvien, 2022-12-05:
> That might be because "everyone" else, like me, have been using AlDente for more than a year now without any such effect. But if you feel better if you can blame it on the app, go along with it.

So everything is black and white and problems don't exist unless they don't happen to you? 

Okay.