AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #798: restart issue " Auto run on startup" failure

Kampi91, 2022-10-24:
**Describe the bug**
A clear and concise description of what the bug is.

Aldente setting enabled "auto run on startup" but, the aldente does not run normally when macbook is restarted. 
Auto-start must be activated on startup, but this is a serious error in battery program.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 - macOS version (https://support.apple.com/en-us/HT201260)
 - AlDente Pro or AlDente Free?
 - AlDente Version?
 - List of all enabled/disabled settings in AlDente:

**Additional context**
Add any other context about the problem here.

hayden0125, 2022-10-24:
It does not appear in application folder when I install it. Auto run on start up either. Should install it again and again whenever Mac restarts. Regret purchase of Pro version.

MatthiasKerbl, 2022-11-01:
> **Describe the bug** A clear and concise description of what the bug is.
>
> Aldente setting enabled "auto run on startup" but, the aldente does not run normally when macbook is restarted. Auto-start [...]

Hi @Kampi91,

We are currently aware that Launch at login does not work for some users.
At the moment, we are trying to find the cause of this issue and hope that we can fix it as quickly as possible.

If you have any new findings about this issue, let me know.

I will also give updates about this here if anything new comes up.

Cheers,
Matthias

MatthiasKerbl, 2022-11-01:
> It does not appear in application folder when I install it. Auto run on start up either. Should install it again and again whenever Mac restarts. Regret purchase of Pro version.

Hi @hayden0125,

Did you follow the installation guide? https://apphousekitchen.com/installation-guide/

Cheers,
Matthias

DamonHD, 2022-11-01:
FWIW, launch at login *started* working for me with Ventura 13.0 (on M1).

Rgds

Damon

MatthiasKerbl, 2023-01-01:
Hi @Kampi91,

We have just released AlDente 1.20 which has a fix for the launch at login issue.

Cheers,
Matthias

closed 2023-01-01 by MatthiasKerbl

darelldd, 2023-02-27:
I have 1.20. Running Ventura 13.2.1.

Approximately every other time that I reboot, Al Dente does not automatically restart. It DOES ask me for my system password though, saying that Al Dente needs to make changes. But adding my password and clicking enter just clears the app, and it ends up not running after the reboot.

MatthiasKerbl, 2023-02-27:
> I have 1.20. Running Ventura 13.2.1.
>
> Approximately every other time that I reboot, Al Dente does not automatically restart. It DOES ask me for my system password though, saying that Al Dente needs [...]

Hi @darelldd,

Sorry for the inconvenience. A fix for this bug should be included in the next update.

Cheers,
Matthias