AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #800: Can't change to clamshell mode

fandiluo, 2022-10-25:
**Describe the bug**
After upgrading macOS Ventura version
Can't change to clamshell mode

**To Reproduce**
1. Click Discharge

**Expected behavior**
Change to clamshell mode

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook model MacBook Pro 16-inch 2019
 - macOS version Ventura 13.0
 - AlDente Free
 - AlDente Version 1.19.1
 - List of all enabled/disabled settings in AlDente: Free Version

MatthiasKerbl, 2022-10-25:
Hi @fandiluo,

Discharge during clamshell mode is only supported in AlDente Pro.

Cheers,
Matthias

fandiluo, 2022-10-25:
Hi @MatthiasKerbl 

But I haven't updated macOS (Ventura) when I can click to discharge.
"discharge", I thought this was a free feature, I don't know how I used to use
[image]

If this is really not a bug, I will consider buying it!

Thanks!

fandiluo, 2022-10-25:
Addendum.
I came home and used another mac (12.6) and it functioned fine

MatthiasKerbl, 2022-10-25:
Hi @fandiluo,

I don't know if I understood your question correctly.

In general, Discharge is a feature that's also available in AlDente Free. However, if you want to use the Discharge feature while your MacBook is in clamshell mode, then you need AlDente Pro.

Cheers,
Matthias

fandiluo, 2022-10-25:
Hi @MatthiasKerbl  

I'll try to explain again
AlDente Free discharge works in 12.6
AlDente Free discharge does not work in 13.0 and shows that you have to upgrade to Pro to use it
If you have macOS 13, I hope you can use the free version to reproduce 
Thank you very much!

closed 2022-10-25 by fandiluo