AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #801: Could you hide the icon in the dock?

martianmartian, 2022-10-25:
hi there, 

thanks for this life saving app. 

just a little request. Could you hide the icon in the dock?  because this is a background app, it makes sense to save some docking room by removing the icon while the app is running.

MatthiasKerbl, 2022-10-25:
Hi @martianmartian,

Due to some people having trouble finding AlDente, we implemented the dock icon and have it activated automatically after installation. However, since most people don’t want to use the dock icon in the long term, we also included a setting to hide it. Just into settings in AlDente -> Appearance -> Other -> and disable Show Dock Icon.

Cheers,
Matthias

closed 2022-10-25 by MatthiasKerbl

GIT-Preserved, 2025-02-10:
Thank you so much, I think it's the fact that Appearance -> Other -> disable Show Dock Icon happens to be perfectly hidden in the settings under the default window size of AlDente that led me here 😂