AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #802: Could you lower the price  please?

martianmartian, 2022-10-25:
Really like your app. 

it's just... the only feature i use is charging limit. would like to contribute financially but not really into other features for 20 bucks...

how about 5 bucks for just charging limit, and discharge, with no ads?

MatthiasKerbl, 2022-10-25:
Hi @martianmartian,

I am glad you like our app AlDente.
If you want to support the development of AlDente but don't want to buy the lifetime license, you can also buy the subscription license for less than half the price.
AlDente Free is completely free and has no advertisements and offers exactly the features you mentioned, Charge Limiter and Discharge.

Cheers,
Matthias

martianmartian, 2022-10-28:
I understand and appreciate your hard work making this app. Just a curious discussion. hope you don't mind. 

Seems to me that keeping track of all the monthly fees burdens me, feels like leeches on my wallet ... i'm also working on an app, so i avoided this type of charging mechanism. 

Seems like more reasonable to have this feature install on mac for once and not have to think about it again. what makes you think people would want to pay you monthly?

asoksevil, 2022-11-06:
Is there any promotion going on anytime soon? I like the app however I also struggle to justify 20 for basically for the charging limit (life time license, not a fan of subscription either)

Many thanks

Bachsau, 2022-11-11:
> what makes you think people would want to pay you monthly?

What makes you think you have a right to get everything for free? It is an option. You can either pay 20 $ and be done with it, get a subscription, or be satified by what the free version offers. Developing software is hard work.

romancin, 2022-11-22:
Are you applying a discount this year for black friday like you did last year? Thank you!

MatthiasKerbl, 2022-12-24:
Hi everyone,

Since this issue turned into a discussion about the price of software in general, I will convert it into a discussion.

Cheers,
Matthias