AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #804: Please provide another payment method besides Paddle or remove it at all.

Volodymyr-13, 2022-10-25:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I see any app which uses Paddle. 
Currently I have 3 apps on my list, which uses Paddle for premium and I'll not buy them due to this bad service. 

**Describe the solution you'd like**
Just Paypal or something.. 

**Describe alternatives you've considered**
Paypal.

**Additional context**
I just wanna pay for lifetime version now and I'll not pay at all if this will be a Paddle = dealbreaker.

Check out how many people already got into trouble with it: https://github.com/davidwernhart/AlDente/issues?q=is%3Aissue+Paddle+

And this is not counting a lot more people, who didn't report this issue here.

MatthiasKerbl, 2022-10-28:
Hi @Volodymyr-13,

I am understanding your frustration.
Unfortunately, it is not as simple as just using PayPal since Paddle is not just our payment provider but is selling our software for us. This means that Paddle is not only taking care of establishing many different payment methods for many currencies but also does taxes in all countries we are selling to. If we would just use PayPal, we would need to have tax consultants in each and every country we want to sell in, and we don't have the recourses to do that. However, we always want to improve, and that's why we always forward all the issues to Paddle so that they can improve. Furthermore, I might start looking into alternatives that we can offer more options for our customers to get AlDente Pro.

Hope that helps to understand the situation better.

Cheers,
Matthias

Volodymyr-13, 2022-10-28:
Hey @MatthiasKerbl check what https://www.codeandweb.com/store/texturepacker-single uses and how it works overall.
I purchased with zero hassle, restores keys if needed, unlinked devices. All was always flawlessly. 

For now, I have only 2 months left for PRO version... after that, without a new payment method, I will no longer be your customer.

MatthiasKerbl, 2022-10-28:
> Hey @MatthiasKerbl check what https://www.codeandweb.com/store/texturepacker-single uses and how it works overall. I purchased with zero hassle, restores keys if needed, unlinked devices. All was [...]

Thanks, @Volodymyr-13, I will check it out.

Volodymyr-13, 2023-01-21:
https://fastspring.com
https://payproglobal.com
https://www.2checkout.com/global-payments/

Volodymyr-13, 2023-01-26:
@MatthiasKerbl 

and today paddle tried to use my money! 
Without any prior e-mail about subscription, like in 1 month will be charge… in 1 week will be charge…
Those bad guys tried silently to stole my money from my bank account!
Only after payment failed, they sent me e-mail telling so and asking to update card! No no no!

Deal breaker. I’m not user of Al Dente Pro anymore.

MatthiasKerbl, 2023-01-27:
> @MatthiasKerbl
>
> and today paddle tried to use my money! Without any prior e-mail about subscription, like in 1 month will be charge… in 1 week will be charge… Those bad guys tried silently to stole [...]

Hi @Volodymyr-13,

You currently have three options to get AlDente Pro and a fourth one is coming soon. The current options are:
1. Our website (through Paddle): https://apphousekitchen.com/pricing/
2. Included in Setapp: https://setapp.com/apps/aldente-pro
3. As a bundle on MacFinder: https://macfinder.co.uk/

We discussed the option of changing our merchant of record internally but decided against it at the moment since the cons outweighed the pros for us. However, it is a continuously changing field and we will reassess it regularly.

Volodymyr-13, 2023-01-27:
1-2 is the same lol. Your site is doing purchase via paddle.
Setapp is a paddle. I’ll never use setapp or paddle, same trash..
3. Direct link to buy? I don’t like word bundle.. should be just this app.