AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #805: Active Discharge in Clamshell Mode is only available with the Pro version

Volodymyr-13, 2022-10-25:
**Describe the bug**
Active Discharge in Clamshell Mode is only available with the Pro version - but I have PRO version.

**To Reproduce**
Steps to reproduce the behavior:
1. Update macOS with software update.
2. System restarts and after that I saw this error message.
3. Anyway discharge works a bit later.

**Expected behavior**
No any error messages, since I have a PRO version. 

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook 2021 M1
 - macOS version 13
 - AlDente Pro
 - AlDente Version 1.9.1
 - List of all enabled/disabled settings in AlDente:

[image]
[image]

Volodymyr-13, 2022-10-25:
Hmm, it's now actually showing me this each Mac restart, but then discharge works.

[image]

Ventura 13.1

MatthiasKerbl, 2022-10-26:
Hi @Volodymyr-13,

Thank you for pointing out this bug.
We could already find the cause of this issue and will fix it with the next update.

Cheers,
Matthias

MatthiasKerbl, 2022-11-02:
Hi @Volodymyr-13,

We have just released AlDente Pro 1.19.2 which has a fix for the bug you mentioned.

Cheers,
Matthias

closed 2022-11-02 by MatthiasKerbl

hossam-96, 2025-02-03:
The bug is still there for version 1.29, it happened after I used Migration Assistant to transfer my data to another Macbook.
I tried updating to 1.30 version, but it's still showing the message.

I don't have pro version, but discharging was working normally before.