AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #806: Automatic Discharge blocking from Sleep/Lock Screen

Volodymyr-13, 2022-10-25:
**Describe the bug**
When Automatic Discharge active, like I have 85% and charge limit is set to 80%, then sleep is disabled. 
And with it, Lock Screen disabled too.

**To Reproduce**
Steps to reproduce the behavior:
1. set options like on a screenshoot
2. use MacBook with external monitor
3. notice that if MacBook is idle it's not been locked or goes to sleep, it just automatically discharges and wait until 80%  

**Expected behavior**
Allow to sleep when Disable Sleep until Charge Limit disabled.
Allow Lock Screen to engage in any case. 

**Screenshots**

[image]
[image]

**MacBook (please complete the following information):**
 - MacBook Pro M1 2021
 - macOS version 13.1
 - AlDente Pro
 - AlDente Version 1.19.1
 - List of all enabled/disabled settings in AlDente:

MatthiasKerbl, 2022-11-05:
Hi @Volodymyr-13,

Unfortunately, we were so far not able to get Discharge or Automatic Discharge to work in the clamshell mode without deactivating sleep for the period of discharging. Furthermore, while sleep is deactivated, Lock Screen is also automatically deactivated with it.
We will try to improve AlDente Pro in this regard by working on getting Lock Screen to work while sleep is disabled and trying to not need to disable sleep completely during Discharge in clamshell mode. However, we are not sure if this is possible at all.

Cheers,
Matthias

jalexvig, 2023-10-19:
Any update on this? Clamshell is 85% of my MBP usage so it's pretty frustrating to not have this working (particularly when I've topped up or aldente is calibrating). Also am I missing something? I don't see anything about this in the online docs (either now or before I purchased pro).

sidewinder666, 2024-11-30:
Pro user here as well, and disappointed to see there's still no resolution to this issue. Same situation as the previous reply, though I use clamshell almost 100%. Has there been _any_ progress on this?

MatthiasKerbl, 2024-12-01:
Hi everyone,

Unfortunately, it is quite difficult to make improvements for this use case due to technical constraints in place from Apple. However, we might have found a path to make some improvements in the upcoming updates.

Best,
Matthias

MatthiasKerbl, 2025-02-21:
Hi everyone,

We have just released AlDente 1.31. which is now considering the Lock Screen and Screensaver settings of macOS when sleep is disabled by AlDente.

Best,
Matthias

sidewinder666, 2025-02-21:
I can confirm that, on my Macbook, screen saver now works correctly when discharging. 

**Good job, dev team!**

sidewinder666, 2025-02-21:
FOLLOWUP:
Uh oh... with latest MacOS update, Sequoia 15.3.1, Screen saver does start at specified time, but when it reaches the display "Off" time, it turns off the external monitor (and the native screen; when I open the lid I see it powerup)... but then will not turn the external monitor back **on**. I have to open the lid so that the native display turns on, **then** the external monitor will turn on.

Has anyone else had this happen?

MatthiasKerbl, 2025-02-24:
> EDIT: The issue I posted yesterday just... went away. I'm baffled why, but it was a couple reboots later that everything just started working as (I think?) it should.
> 
> **Screen saver does start after the set time, when discharging.**
> 
> But it does not turn off the display when discharging. (not "Sleep"... "Display Off".)
> 
> Yesterday, it did turn off the display, but then would not wake up the external screen unless I opened the clamshell first.
> 
> **Is that the correct behavior though? That it does NOT turn off the display(s) when discharging?**

Hi @sidewinder666,

Good to hear that everything is working now as intended.

Best,
Matthias

closed 2025-02-24 by MatthiasKerbl