AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #807: Clamshell Discharging shuts down USB-C Dock & Connections

Sebbo02, 2022-10-26:
**Describe the bug**
I am using an HP USB-C Docking Station with my MacBook Air M1. After using the Free Version for many months, I just purchased the PRO Version to Discharge my MacBook while sitting in Clamshell-Mode on my Desk. Unfortunately this seems to not work as AlDente not only cuts down the power supply, but the whole USB Connection (Monitor go off). Is this a normal behavior with a Docking Station? How to fix this? 

**To Reproduce**
Steps to reproduce the behavior:
1. Having 100% Charge level & connected to Docking Station, Go to 'AlDente'
2. Click on 'Discharge' to 80%
3. Wait a couple of seconds
4. See error (Monitors go off)

**Expected behavior**
I expect that AlDente stops charging the battery, but not deactivating USB-C

**MacBook (please complete the following information):**
 - MacBook Air 2021 M1
 - macOS version Ventura 13.0
 - AlDente Pro 1.19.1

Barthcbb, 2022-10-29:
I have a similar issue with a USB hub. When AlDente is running, the connection with external hard drives through the USB hub is very problematic—the disks keep losing connection. So whenever I want to use external drives with the USB hub, I have to quit AlDente for that moment.

MatthiasKerbl, 2022-11-03:
Hi @Sebbo02 and @Barthcbb,

We have just released AlDente Pro 1.19.2 which has a couple of tweaks in regard to Discharge during the clamshell mode.
Let me know if it improved any of the issues you experienced.

Cheers,
Matthias

Sebbo02, 2022-11-04:
Hi @MatthiasKerbl . Many thanks for taking care. But unfortunately the new version didn't fix the issue. 

I installed 1.19.2 with 2% Battery. Set charge limit to 80% and "stop charging while sleep" to on. After a couple of hours my battery was at 100% so for whatever reason the limit was ignored. Nevertheless this was a good opportunity to test the discharge function again. But unfortunately same behavior as mentioned above. Hope you find some hint in my debug file.

[aldentepro-debug-2022-11-04-21:02:39.zip](https://github.com/davidwernhart/AlDente/files/9941626/aldentepro-debug-2022-11-04-21.02.39.zip)

MatthiasKerbl, 2022-11-06:
Hi @Sebbo02,

Thanks for reporting back.
The Charge Limiter should definitely work as intended. Could you send me a screenshot of the Charge Settings page in AlDente?
Also, I am currently trying to replicate the issue you encounter with USB connections disconnecting during Discharge in Clamshell mode.
I have a couple of USB devices plus a monitor through HDMI plus power connected to a USB-C dongle which is plugged into my MacBook with a single USB-C connection. When I activate Discharge during clamshell mode, everything is working as intended. My MacBook starts discharging, all USB devices still work without any interruption, and the connection to my external monitor does also not get interrupted.
Which dock are you using?
Do you have a second dongle/dock you could try?

Cheers,
Matthias

MatthiasKerbl, 2022-11-07:
Hi @Sebbo02,

After reviewing your debug file in more detail, it seems like macOS is in some kind of safety state which doesn't allow AlDente to make any changes to the charging behavior. Usually, this happens after macOS updates.
Please try the following steps to get macOS out of this safety state:

> To get AlDente to work again, please try the following steps which have worked for many users before:
> 1.  Unplug and discharge your MacBook to less than your current charge limit in AlDente
> 2. Turn off your MacBook and close the lid for a couple of minutes
> 3. Unplug the power brick from the outlet and wait a couple of minutes
> 4. Turn on your MacBook and plug it in again
> 5. Check if it stops charging at the charge limit set in AlDente

Please report back when you have tried it.

Cheers,
Matthias

Sebbo02, 2022-11-07:
Hi Matthias,

indeed it seems that this was the issue. I followed above steps and now it seems to work fine again. Always surprising to find out what interesting behavior macOS does in the background. 

Many thanks for this hint!

MatthiasKerbl, 2022-11-07:
Hi @Sebbo02,

I am glad to hear that.
Does this also mean that Discharging with the USB-C Dock works now as well?

Cheers,
Matthias

Sebbo02, 2022-11-07:
Yes "discharging" as well as "stop charging in sleep" works fine with the Dock. Maybe my MacBook didn't enter all deep-sleep stages yet, but as of now I am very confident that everything works as it should. In case I still have issues, I will let you know. 

Many thanks again :-)

closed 2022-11-07 by MatthiasKerbl