AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #809: Can't use Aldente free with OSX Ventura (13.0)

mabnz, 2022-10-27:
I used to use Aldente with 12.6, and it worked fine, discharging to the set limit then keeping the charge level capped at the amount I set.

Upon upgrading to Ventura, this discharge does not seem to work.

mabnz, 2022-10-27:
I should note, I have the Mac lid closed (Mac M2). Whilst I can see "active discharge" is limited to Pro, it seems even "passive discharge" doesn't work with the lid closed on Aldente 1.19.1 with OSX 13.0.

MatthiasKerbl, 2022-10-27:
Hi @mabnz,

I am not 100% sure if I understand your issue completely.
In general, the Discharge feature during clamshell mode is not supported in the free version. However, if you are not in clamshell mode, Discharge should work fine.

- Does Discharge work for you when your MacBook is not in clamshell mode?
- Did Discharge work for you even in clamshell mode on macOS 12.6?

Cheers,
Matthias

mabnz, 2022-10-27:
> * Does Discharge work for you when your MacBook is not in clamshell mode?
> * Did Discharge work for you even in clamshell mode on macOS 12.6?

Hi @MatthiasKerbl,

That's correct. When I say "discharge" i'm referring to the feature where the Mac won't charge even though the power is plugged in. Charge is at 92%, it will prevent charging until the charge point is reached. This isn't happening in these two cases:

- When using macOS 13.0 at all (used to work in 12.6)
- When the lid is closed (used to work in 12.6)

I'm not referring to the "active discharge" feature (which however, also isn't working).

I should note, the Mac is being charged via USB-C from my monitor. Will that make any difference?

Cheers,
Mike

mabnz, 2022-10-27:
Update: Strangely, after a few reboots it's now working in both OSX 13.0 with the lid closed.

MatthiasKerbl, 2022-10-30:
> Update: Strangely, after a few reboots it's now working in both OSX 13.0 with the lid closed.

Hi @mabnz,

Thank you for the update.
Do I understand correctly that Discharge is now working for you with the lid open and the lid closed?

Cheers,
Matthias

mabnz, 2022-10-31:
> Thank you for the update. Do I understand correctly that Discharge is now working for you with the lid open and the lid closed?

Hi @MatthiasKerbl, yes that's correct. It's working in both situations.

closed 2022-11-01 by MatthiasKerbl

t5a5ak1, 2022-11-02:
Abosolutely same situation.
Upgrade to Ventura,after reboot, the ploblem is gone.

Betristor, 2022-11-03:
Same issue, and the reboot solved everything.