AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #810: AlDente under Ventura does not work

dnns123, 2022-10-27:
Hello,

I have updated to Mac Ventura today. Unfortunately, since then AlDente no longer works. Even though it is set to 80%, the laptop loads to 100%.

Does anyone have a similar error?

Thanks

MatthiasKerbl, 2022-10-28:
Hi @dnns123,

This might be due to a known bug. Please check out this comment I made in another thread:

        Hi everyone,

We are sorry that so many of you had issues with AlDente after the newest macOS update.
We are aware of this issue but did not find a solution yet. However, we will keep working on it and give updates here.
I have also updated our FAQs about this topic:

_**AlDente stops working after the macOS update!**_

_Unfortunately, AlDente temporarily does not work after a macOS update for a small number of users._

_To get AlDente to work again, please try the following steps which have worked for many users before:_

_1. Unplug and discharge your MacBook to less than your current charge limit in AlDente_
_2. Turn off your MacBook and close the lid for a couple of minutes_
_3. Unplug the power brick from the outlet and wait a couple of minutes_
_4. Turn on your MacBook and plug it in again_
_5. Check if it stops charging at the charge limit set in AlDente_

_We are aware of this bug for a while now but were so far not able to fix it. However, we will keep trying._

One possibility to actively avoid this issue during the next macOS update is to close AlDente completely before the update and restart it when the update is finished.

Cheers,
Matthias

_Originally posted by @MatthiasKerbl in https://github.com/davidwernhart/AlDente/issues/768#issuecomment-1252390562_

r0kas, 2022-11-04:
Having the sam issue, will try your suggestion once I'll get my battery down from unwanted 100%.
My machine is 16-inch 2021 MacBookPro. M1 Pro chip with 32gb ram.

woker001, 2022-11-06:
AlDante abnormally shuts down when in sleep mode in clamshell. (M2 MacBook air 16GB , OS 13.0)

MatthiasKerbl, 2022-11-08:
Hi @dnns123,

Any updates on this?
Did you try my suggested fix?

Cheers,
Matthias

ashraf-soliman, 2022-11-12:
> Hi @dnns123,
> 
> Any updates on this? Did you try my suggested fix?
> 
> Cheers, Matthias

I tried the steps you mentioned, and it WORKED!
Thanks a lot.

vsisl, 2022-11-14:
> To get AlDente to work again, please try the following steps which have worked for many users before:
>
> 1. Unplug and discharge your MacBook to less than your current charge limit in AlDente
> 2. Turn [...]

I tried this and it didn't help. 

Setup: Intel 16" MacBook Pro 2019, macOs Ventura 13.0.1 (22A400), charging through HP U28 HDR display.

MatthiasKerbl, 2022-11-14:
> I tried this and it didn't help.
>
> Setup: Intel 16" MacBook Pro 2019, macOs Ventura 13.0.1 (22A400), charging through HP U28 HDR display.

Hi @vsisl,

What exactly is the issue?

Cheers,
Matthias

MatthiasKerbl, 2022-11-24:
Hi @dnns123,

Hope you could resolve this issue with the instructions above.
I will close this issue for now but feel free to report back if you still experience issues.

Cheers,
Matthias

closed 2022-11-24 by MatthiasKerbl

pseast, 2022-12-09:
I have also had the issue where aldente closes by itself with Ventura 3.0. I have set all permissions to run in background as far as I can tell. It worked before updating.Just tried the above, will see if it helps.

pseast, 2022-12-13:
But I did try the above and it seems to be working again. What do you know?!