AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #813: saling mode disabled battery calibration....

gselsidi, 2022-10-29:
I had battery calibration on, then i turned on sailing mode, and it disabled the calibration have to start all over again...... Please change this behavior.

MatthiasKerbl, 2022-10-30:
Hi @gselsidi,

I understand your frustration about the aborted Calibration Mode.
It's the first time that someone brings this up since usually, our users activate the Calibration mode, let it run, and then change the charge settings again.
I will put it on our list for future improvements but I hope you understand that it is not a priority at the moment.

Cheers,
Matthias

gselsidi, 2022-10-30:
Yeah it’s fine not a big issue thanks

Bachsau, 2022-11-11:
I also think it would just be more clear, if this was as special mode, that, while active overrides all other settings. You should also apply this change to the "Top Up" mode. Currently it moves the slider to 100% while active.

MatthiasKerbl, 2024-06-18:
Hi @gselsidi,

We have just released AlDente 1.27 which now allows you to change the Sailing Range and Charge Limit while the Calibration Mode is active.

Best,
Matthias

closed 2024-06-18 by MatthiasKerbl