AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #816: Battery charge level drops during sleep

shimo-web, 2022-10-31:
Hi, what could be the reason that if I unplug the cable overnight from my MacBook Pro Mid 2015 (manufactured in 2018, coconutBattery shows 95% capacity and 270 cycles) then from the 80% percentage that AlDente Pro 1.19.1 maintains I am left with 45% when I open the computer in the morning? Thank you in advance for your guidance.

shimo-web, 2022-10-31:
My settings:
[image]
[image]

MatthiasKerbl, 2022-10-31:
Hi @shimo-web,

It seems like your MacBook does not go to sleep. The question is why.
Could you generate and share a debug file? https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/
Do you use any other app that controls the sleep behavior of your MacBook, such as Amphetamine or Lungo?

Cheers,
Matthias

shimo-web, 2022-10-31:
No, no such apps. However I checked Activity monitor to find that nesessionmanager process is using 99-100% CPU. I think it was related to VPN app I'm testing (Hotspot Shield). Turned it off and it's gone. Anyway, here goes the debug file. I have added .txt as Github didn't let me upload .pasta. :)
[aldentepro-debug-2022-10-31-11:08:56.pasta.txt](https://github.com/davidwernhart/AlDente/files/9900143/aldentepro-debug-2022-10-31-11.08.56.pasta.txt)

MatthiasKerbl, 2022-11-01:
> No, no such apps. However I checked Activity monitor to find that nesessionmanager process is using 99-100% CPU. I think it was related to VPN app I'm testing (Hotspot Shield). Turned it off and [...]

Hi @shimo-web,

Unfortunately, we could not see any issues in the debug file.
Could you please try to completely uninstall AlDente and check if you still experience this issue?

Cheers,
Matthias

MatthiasKerbl, 2022-11-01:
> Sure I can do that but should it be done any special way or simply delete the app?

Please follow the this guide to uninstall AlDente: https://apphousekitchen.com/how-to-uninstall-aldente-pro/

MatthiasKerbl, 2022-11-15:
Hi @shimo-web,

Any updates on this?
Did you follow our how-to-uninstall guide and test it again?

Cheers,
Matthias

MatthiasKerbl, 2022-12-24:
Hi @shimo-web,

Since you have not reported back in a long time, I will close this issue for now. However, feel free to get back and I will open the issue again.

Cheers,
Matthias

closed 2022-12-24 by MatthiasKerbl