AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #822: Set charge limit based on voltage

stendro, 2022-11-09:
Since there's a problem with the SoC drifting over time, why not have the option of setting a voltage threshold. 

There's a lot of information available from the battery:
`ioreg -w 0 -f -r -c AppleSmartBattery | grep Voltage`

On my macbook m1 air I see there's a 3 cell battery, each cell having a nominal voltage of about 3.8v, and about 4.4v when full. So it would not be difficult to determine what is an appropriate threshold to set. Right now it's showing 47% and a voltage of 11,426 (11,426/3 = 3.8).

Why do I want this? I'd like to avoid regular battery re-calibrations. If I'm using it docked all the time I don't want the extra hassle.

I don't actually have the pro version. With this feature it would be much more appealing. Right now I'm not sure Pro would make any difference to me.

Bachsau, 2022-11-11:
The biggest difference of Pro is that you have the "sailing" function available. So instead of trying to keep your battery at a certain level, you can set a range in which to leave it alone, which also means you won't have to re-calibrate. If your MacBook is docked all the time, it will just run on grid power and it will take weeks for the battery to reach the lower threshold at which Al-Dente will decide to charge it again.

stendro, 2022-11-13:
> The biggest difference of Pro is that you have the "sailing" function available. So instead of trying to keep your battery at a certain level, you can set a range in which to leave it alone, which also means you won't have to re-calibrate. If your MacBook is docked all the time, it will just run on mains current and it will take weeks (maybe months) for the battery to reach the lower threshold at which Al-Dente will decide to charge it again. Of course you could also do that manually by charging it to, say 70%, then set your maximum to 60%. As long as you don't press the discharge button, nothing will happen.

That is not what I meant. Purportedly, the battery % gets out of sync with the actual battery voltage after some time. And I do seem to have experienced this after first installing AlDente, but not again since that first instance.
It seemed to happen because I was not using the macbook then, it was just sleeping/powered off. Then one day it shut of at around 20%. This was before I knew how to check the battery info via terminal.

Now I'm using the mac quite a bit, and observing the voltage shows that it does stay at the expected level.