AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #825: AlDente does not block the load at 80%

Nemrod7, 2022-11-17:
**Describe the bug**
AlDente does not block the load at 80%. In fact, after installation, I let my 2018 MBP 15 discharge below 80% and then plug it back in, some time later it is at 100%. I also have the same issue on my 2019 MBP13.

I use a Caldigit TS3 plus.

**To Reproduce**
Steps to reproduce the behavior:
1. Lacune AlDente
2. Check that 80% is set as the limit
3. Unplug my MBPs until the charge is under 80% is plug them back to power

**Expected behavior**
I expect my battery to be at 80%

**Screenshots**
https://ibb.co/f8n65px
https://ibb.co/LZgYnjh
https://ibb.co/M6tWHXD

**MacBook (please complete the following information):**
 - MacBook 15 2018 & 13 2019
 - macOS v12.6
 - AlDente Free
 - AlDente  1.19.2
 - List of all enabled/disabled settings in AlDente: please see captures

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2022-11-17:
Hi @Nemrod7,

Did you do a macOS update lately?

Cheers,
Matthias

Nemrod7, 2022-11-17:
Hi Mattias,

Yes, but that was the case before the last update. I even reinstalled macOS.

Bests

MatthiasKerbl, 2022-11-17:
Hi @Nemrod7,

Thanks for the update.
I would still recommend the following steps since it might have something to do with this bug #768:

> To get AlDente to work again, please try the following steps which have worked for many users before:
> 
> 1. Unplug and discharge your MacBook to less than your current charge limit in AlDente
> 2. Turn off your MacBook and close the lid for a couple of minutes
> 3. Unplug the power brick from the outlet and wait a couple of minutes
> 4. Turn on your MacBook and plug it in again
> 5. Check if it stops charging at the charge limit set in AlDente

Let me know if it worked.

Nemrod7, 2022-11-18:
Hi @MatthiasKerbl,

Merci, je teste.

Bests

Nemrod7, 2022-11-18:
Hi @MatthiasKerbl,

Thanks, it's working on my MBP15, so I guess it will work on my MBP13.

Bests

MatthiasKerbl, 2022-11-19:
Hi @Nemrod7,

Thanks for reporting back and I am glad it works now.
I will close this issue for now but let me know when you run into any issues with the other MacBooks.

Cheers,
Matthias

Nemrod7, 2022-11-20:
Hi @MatthiasKerbl,

I've just checked, my battery is at 94%.

Best

MatthiasKerbl, 2022-11-21:
> Hi @MatthiasKerbl,
> 
> I've just checked, my battery is at 94%.
> 
> Best

Hi @Nemrod7,

are you talking about the 13-inch model?

Nemrod7, 2022-11-21:
Hi @MatthiasKerbl,No, the 15.Best Sent from a phone that often changes words I meant to say for words I did not intend.

MatthiasKerbl, 2022-11-21:
Hi @Nemrod7,

What was the exact scenario when it charged above the Charge Limit?
Did you do anything different compared to when it worked before?
Did you close the lid, did your MacBook go to sleep or did you turn it off in the meantime?

Nemrod7, 2022-11-24:
Hi @MatthiasKerbl,

I proceeded with the manipulations you specified, I looked at the battery level shortly after, I was well at 80%, then I continued to use my MBP15 as usual then I saw that I was at 94% and now 100%.

Best

MatthiasKerbl, 2022-11-24:
Hi @Nemrod7,

Did you close the lid, did your MacBook go to sleep or did you turn it off while you used it as usual?

Nemrod7, 2022-11-24:
@MatthiasKerbl,

I rarely close the lid, not recently, but yes, it went to sleep as usual.

MatthiasKerbl, 2022-11-24:
> @MatthiasKerbl,
> 
> I rarely close the lid, not recently, but yes, it went to sleep as usual.

This is most likely why it continued charging.
This is from our FAQs about this topic:

> AlDente Free can not control the charging behavior of your MacBook while it is in sleep or shut down. Therefore, it will continue to charge to 100% when you close the lid or shut it down. AlDente Pro has features called “Stop charging when sleeping” and “Stop charging when powered off” to counteract this behavior. You can read more about these features in our blog:
> https://apphousekitchen.com/feature-explanation-stop-charging-when-sleeping/
> https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/

Nemrod7, 2022-11-26:
@MatthiasKerbl,

I would like to check one point, what do you mean by sleep, in fact my Mac is in standby, I'm not sure if it's the same thing and it's the normal operationa: use > screensaver > black screen.

Best

MatthiasKerbl, 2022-11-26:
> @MatthiasKerbl,
> 
> I would like to check one point, what do you mean by sleep, in fact my Mac is in standby, I'm not sure if it's the same thing and it's the normal operationa: use > screensaver > black screen.
> 
> Best

Hi @Nemrod7,

With sleep, I mean the sleep mode Macs have which is basically the standby mode of Macs but it is called sleep within macOS. Sleep is usually activated by closing the lid of MacBooks, manually by clicking on the apple icon in the upper left corner and on sleep.

Cheers,
Matthias

MatthiasKerbl, 2022-12-05:
> @MatthiasKerbl,
> 
> So my Mac is very rarely in sleep mode, so I should be able to limit the charge to 80%?
> 
> Best

Hi @Nemrod7,

Yes, AlDente should definitely be able to limit charging on your devices.

Cheers,
Matthias

Nemrod7, 2022-12-05:
@MatthiasKerbl,

Thanks for your answer, but then what did I miss, what should I do to make it work?

Best

MatthiasKerbl, 2022-12-05:
> @MatthiasKerbl, Thanks for your answer, but then what did I miss, what should I do to make it work? Best
> […](#)

To be honest, I don't really know why it does not work for you. Your devices are definitely supported and AlDente usually works within its limitations on those devices.
Maybe it would help to start from zero again.

- Uninstall AlDente completely following this guide: https://apphousekitchen.com/how-to-uninstall-aldente-pro/
- Download and install AlDente again: https://apphousekitchen.com/installation-guide/
- Deactivate Apple's optimized battery charging.
- Set charge limit in AlDente to 80%
- Unplug and discharge your MacBook below 80%
- Plug in your MacBook again and check if the charge limiter is working

Cheers,
Matthias

stgmsa, 2023-01-26:
I'm having the similar issue, but mine is: it charges just a little bit more than the limit. 

Here is my scenario:
1. charging threshold set to 80%, while the battery level is at about 30%.
2. plug the USB-C Cable (18W)
3. it starts charging slowly (because of the 18w power adaptor) 
4. left the Mac idle for couple of hours - then it should turn off the display after 15 minutes of idling and it did turned off the display automatically after about 20 min.
5. I dont know if it will go to sleep automatically (the sleep mode you mentioned - like somehow manually triggered by clicking "apple menu" -> "sleep")
7. In the morning of next day, I noticed the battery level is 83%, clicking on the battery icon , shows "battery not charging"

Here is some of the battery settings:

"system preference" -> "battery" -> "Power adaptor"
1. prevent your mac from automatically goto sleep while display is off - unchecked
2. wake up for internet access - unchecked
3. low power mode - unchecked  

My expection is the charging process will stop when it reaches 80%, but there's 3% more than the threshold.

MatthiasKerbl, 2023-01-27:
> I'm having the similar issue, but mine is: it charges just a little bit more than the limit.
> 
> Here is my scenario:
> 
> 1. charging threshold set to 80%, while the battery level is at about 30%.
> 2. plug the USB-C Cable (18W)
> 3. it starts charging slowly (because of the 18w power adaptor)
> 4. left the Mac idle for couple of hours - then it should turn off the display after 15 minutes of idling and it did turned off the display automatically after about 20 min.
> 5. I dont know if it will go to sleep automatically (the sleep mode you mentioned - like somehow manually triggered by clicking "apple menu" -> "sleep")
> 6. In the morning of next day, I noticed the battery level is 83%, clicking on the battery icon , shows "battery not charging"
> 
> Here is some of the battery settings:
> 
> "system preference" -> "battery" -> "Power adaptor"
> 
> 1. prevent your mac from automatically goto sleep while display is off - unchecked
> 2. wake up for internet access - unchecked
> 3. low power mode - unchecked
> 
> My expection is the charging process will stop when it reaches 80%, but there's 3% more than the threshold.

Hi @stgmsa,

Unfortunately, the battery percentage likes to "jump around" a couple of percents sometimes since it is difficult to say what the exact battery level at a certain moment is. To keep it more accurate at the current Charge Limit, you can activate the Automatic Discharge feature.

Cheers,
Matthias

stgmsa, 2023-01-27:
Gotcha. Thank you for explaining.
Just wanna confirm,

Say, typically if I set the charge level to 80% it will stop charging when
battery level reaches around 80%,
then it might "jump around" within 75%-85%,

If I just want the battery never to get charged when the battery is at a
level I can accept, e.g. 70% - 80% ,  even 60%+ is totally fine to extend
the battery life,
Can I set the battery level to 50%, so even if the battery level jumps
around, it won't reach the bottom charging level, my battery wont get
charged for even one second?

MatthiasKerbl, 2023-01-27:
> Gotcha. Thank you for explaining. Just wanna confirm, Say, typically if I set the charge level to 80% it will stop charging when battery level reaches around 80%, then it might "jump around" within [...]

Hi @stgmsa,

When you set the charge limit at 80% it will stop charging when 80% is reached. When it jumps above that later, it will still not charge. When it jumps below that, it will charge to 80% again.

I am not sure if I understand your second question. Could you explain it again in more detail?

stgmsa, 2023-01-27:
I'm doing like this:
1. application not launched, plug the cable, until it charges to 80%.
2. unplug the charger
3. open AlDente, set the charge level to 50% (the battery is now at 80%) and let the AlDente running.
4. plug the charger again (at it will fully support the power comsumption, and the battery level should not drop down below 70% even taken the "battery jump around" issue into consideration)

in the scenaral above , the battery won't get charged at all after I launched the AlDente, right ?

MatthiasKerbl, 2023-01-27:
> I'm doing like this:
>
> 1. application not launched, plug the cable, until it charges to 80%.
> 2. unplug the charger
> 3. open AlDente, set the charge level to 50% (the battery is now at 80%) and let the [...]

Oh okay, now I understand what I mean.

> ...and the battery level should not drop down below 70% even taken the "battery jump around" issue into consideration)

Actually, the battery might still lose some energy slowly, because batteries discharge themselves over time and sometimes the battery might jump in as a buffer for a short period.

> in the scenaral above , the battery won't get charged at all after I launched the AlDente, right ?

Yes, that is correct.

You can also check out the Sailing Mode feature of AlDente Pro. It basically does what you are looking for. You can read more about it on our blog here: https://apphousekitchen.com/feature-explanation-sailing-mode/

MatthiasKerbl, 2023-03-12:
Hi @Nemrod7,

Any updates on your issue?
Are you still experiencing this issue with our newest version AlDente 1.21.1?

Cheers,
Matthias

Nemrod7, 2023-03-12:
Hi @Matthias,

Let me check this evening withe last update.

Best

Nemrod7, 2023-03-12:
On, àa ne fonctionne toujours pas chez moi, je suis passé 81%

Best

MatthiasKerbl, 2023-03-12:
Are you only using your MacBooks with the Caldigit TS3 plus dock or did you try it with the Apple power brick as well?

Nemrod7, 2023-03-13:
Hi @Matthias,

I let Al Dente running since yesterday, I just got back, the battery is at 80%, there must be an adjustment period?

Best

MatthiasKerbl, 2023-03-13:
> Hi @matthias, I let Al Dente running since yesterday, I just got back, the battery is at 80%, there must be an adjustment period? Best
> […](#)

Hi @Nemrod7,

Does the Charge Limiter work now?

Nemrod7, 2023-03-14:
> … <x-msg://11/#>

Hi @matthias,

Like I said yesterday, yes !

Best

MatthiasKerbl, 2023-03-14:
> Hi @matthias, Like I said yesterday, yes ! Best
> […](#)

Hi @Nemrod7,

Sounds good. Also keep in mind that if you set the charge limit to 80%, your MacBook might stop charging between 80-83%. Also, I will close this issue for now but let me know if anything comes up again.

Cheers,
Matthias

Nemrod7, 2023-03-16:
> Hi @matthias <https://github.com/matthias>, Like I said yesterday, yes ! Best
> … <x-msg://1/#>

Hi @MMatthias,

My charge is back to 100%, but maybe it's because I lowered the screen ?

Best

MatthiasKerbl, 2023-03-16:
> Hi @mmatthias, My charge is back to 100%, but maybe it's because I lowered the screen ? Best
> […](#)

Hi @Nemrod7,

I am sorry to hear that it does not work again. What exactly do you mean by "lowered the screen"?

Nemrod7, 2023-03-16:
> … <x-msg://3/#>

@Matthias,

J’ai rabaissé l’écran de mon Mac en partant ce matin.

Best

MatthiasKerbl, 2023-03-29:
Hi @Nemrod7,

Did you try the Intel Mode?

Cheers,
Matthias

Nemrod7, 2023-04-12:
Hi,

Sorry for the delay, everything is ok.

Best

closed 2023-04-13 by MatthiasKerbl