AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #827: A notice from AlDente-Pro shows up when mac restarts but it flashes past so quickly

connor-circle, 2022-11-19:
**Describe the bug**
A notice from AlDente-Pro shows up "sometimes" when mac restarts but it flashes past so quickly. I have tried to record my screen by phone, but the notice pops up randomly, so I haven't seen it until now. 

**To Reproduce**
Steps to reproduce the behavior:
1. Restart Mac 
2. login my account and wait the system load items(including AlDente-Pro)
3. See the notice(sometimes)

**Expected behavior**
At least let me see what it is on the notice 
**MacBook (please complete the following information):**
 - Macbook air M1 2020
 - Ventura 13.0.1
 - AlDente Pro 
 - AlDente Version-1.19.2
 - List of all enabled/disabled settings in AlDente:
[image]
[image]
[image]

**Additional context**
I just want to know what the notice want to inform me of. I have checked the Q&A but nothing refer to it so I think the dear developer should know that.

MatthiasKerbl, 2022-11-21:
Hi @connor-circle,

This might be the warning message that the AlDente helper app is not running. This can happen when your MacBook is just getting powered on and the main AlDente app started slightly before the helper app. Therefore, the main app thinks the helper app is missing but when the helper starts as well only shortly after, the message is gone again. This should not result in any other issues.

Cheers,
Matthias

connor-circle, 2022-11-21:
> Hi @connor-circle,
>
> This might be the warning message that the AlDente helper app is not running. This can happen when your MacBook is just getting powered on and the main AlDente app started [...]

Thanks your detailed reply! Have a nice day!

closed 2022-11-21 by MatthiasKerbl