AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #833: can't start calibration mode after shut down

zynth17, 2022-11-25:
**Describe the bug**
so i'm in the middle of calibration mode on discharge to 15% step, and i'm shut down my mac and turn it back on basically a hard restart.
when my mac start, aldente is still on discharge mode, but when i open the settings the calibration mode button is not activated. so i try to press the button but nothing happens.

**To Reproduce**
Steps to reproduce the behavior:
1. start calibration mode
2. wait until the discharge part (step 2)
3. shut down your mac
4. turn it back on 
5. open al dente settings

**Expected behavior**
calibration mode work normally

**Screenshots**
[image]
[image]

**MacBook (please complete the following information):**
 - MacBook model : MacBook m1 Pro 14 inch
 - macOS version Ventura
 - AlDente Pro or AlDente Free? Pro 
 - AlDente Version? 1.19.2
 - List of all enabled/disabled settings in AlDente:
 
[image]
[image]
[image]
[image]

zynth17, 2022-11-25:
just in case you need the debug file.
[Uploading aldentepro-debug-2022-11-25-10:35:33.pasta.zip…]()

MatthiasKerbl, 2023-01-12:
Hi @zynth17,

In general, currently, the Calibration Mode is not meant to continue working after a restart. You need to restart the Calibration Mode again if you restart your MacBook.
Unfortunately, I am currently not able to reproduce your bug that the Calibration Mode does not work again after the restart. Are you still experiencing this issue with our newest version AlDente 1.20?

Cheers,
Matthias

skrivanos, 2023-01-26:
@MatthiasKerbl  I just experienced this with 1.20. Had a crash while discharging, then couldn't click start calibration again after boot. Clicking the button simply did nothing. Solved it by restarting AlDente a few more times and eventually it worked again.

On another note, please add a clearer button to toggle between the calibration steps or make the steps themselves clickable - took me a while to figure out.

MatthiasKerbl, 2023-09-18:
Hi @hi-reeve,

Noting the extended period since our last communication, I am proceeding to close this issue for the time being. Please remember that should you need to revisit this matter, don't hesitate to reach out. I'll be more than happy to reopen the issue and assist you further.

Best,
Matthias

closed 2023-09-18 by MatthiasKerbl