AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #834: Smaller Menu Item

Mayerhofer-dev, 2022-11-27:
When configuring the menu item to show only the current percentage, its width is still very large compared to other items.

Unfortunately, space in the menu bar is very rare on the new MBP14" with notch.

I would really appreciate an option to change the padding around the symbol.

Thanks in advance
Michael

MatthiasKerbl, 2022-11-27:
Hi @Mayerhofer-dev,

You are right, we will try to fix it with one of the future updates.

Cheers,
Matthias

Crazy365, 2022-11-28:
> Hi @Mayerhofer-dev,
> 
> You are right, we will try to fix it with one of the future updates.
> 
> Cheers, Matthias

Good to know you are working on that, I got the same problem. 

And I hope the percentage's font size can be a little bigger, or can be adjustable.

Thanks.

MatthiasKerbl, 2023-01-01:
Hi @Mayerhofer-dev,

We have just released AlDente Pro 1.20 which has reduced spacing if just the percentage is shown.

Cheers,
Matthias

closed 2023-01-01 by MatthiasKerbl