AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #842: Minus 1 percent capacity in 2 days

cyberde, 2022-12-03:
I decided to use the free version for 2 days to test the functionality of the application and got a 1% loss in capacity. I'm shocked. I set the limit to 70%, then discharged. Then I set it to 60%, simulating the behavior of the "sailing mode". 
1) On what principle does the charge control work, bypassing the standard "smart" option? 
2) What could have caused the loss of capacity in such a short time? 

Number of cycles: 57. 
Current capacity 90%.

Awesome.....

cyberde, 2022-12-03:
There can be no coincidences or "cosmic" anomalies, everything happened after a short time of using your program! For 2 days minus 1 percent, it turns out for 4 days 2 percent, and so on ......

cyberde, 2022-12-03:
PS: i don't use "cocount battery" and etc, i used to system information.

MatthiasKerbl, 2022-12-15:
Hi @cyberde,

> 1. On what principle does the charge control work, bypassing the standard "smart" option?

We reverse-engineered the technical pathways Apple implemented to control the charging behavior and control it through AlDente. Therefore, using AlDente is a low risk since it uses the same mechanisms as macOS.

> 2. What could have caused the loss of capacity in such a short time?

It is a common misconception that the battery capacity reported by macOS is accurately tracking and shows the correct real capacity left in the battery. Unfortunately, determining the capacity of a battery is not as easy and accurate as looking at the fill level of a fuel tank for example. Rather, it is based on a lot of guessing and estimating with a lot of uncertainty. Therefore, the capacity rating fluctuates up and down quite a bit.

It is true that limiting charging to a lower level for a long time without doing full cycles can enhance the uncertainty regarding the battery capacity and lead to a miscalibrated battery. However, that's why we implemented the Calibration Mode in AlDente Pro to keep the battery calibrated and avoid this scenario.

Hope that clarifies some things.

Cheers,
Matthias

cyberde, 2022-12-18:
> Hi @cyberde,
>
>
> We reverse-engineered the technical pathways Apple implemented to control the charging behavior and control it through AlDente. Therefore, using AlDente is a low risk since it uses [...]

I again lost another 1 percent capacity in 2 days…..last time I thought it was a coincidence, but no! Can you give comments or a way to return the lost interest?

MatthiasKerbl, 2023-01-12:
Hi @cyberde,

I hope I could clarify most things in my previous post. Therefore, I will close this issue.

Cheers,
Matthias

closed 2023-01-12 by MatthiasKerbl