AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #848: not working properly on mbp 14 m1 pro

xxMosho, 2022-12-08:
whenever I set a limit it always exceed the limit 
also when I try to discharge it keeps going from to discharge to charging every couple of seconds

MatthiasKerbl, 2022-12-08:
Hi @xxMosho,

Did you do a macOS update recently?

Cheers,
Matthias

xxMosho, 2022-12-08:
yes but that was before using the app

MatthiasKerbl, 2022-12-08:
> yes but that was before using the app

This still might have triggered a known bug #768.
Please try the following steps:

> To get AlDente to work again, please try the following steps which have worked for many users before:
> 
> 1. Unplug and discharge your MacBook to less than your current charge limit in AlDente
> 2. Turn off your MacBook and close the lid for a couple of minutes
> 3. Unplug the power brick from the outlet and wait a couple of minutes
> 4. Turn on your MacBook and plug it in again
> 5. Check if it stops charging at the charge limit set in AlDente

Cheers,
Matthias

xxMosho, 2022-12-08:
it worked fine after I uninstalled and then reinstalled the application with the uninstallation guide your provided on your website
thanks for helping <3

MatthiasKerbl, 2022-12-08:
> it worked fine after I uninstalled and then reinstalled the application with the uninstallation guide your provided on your website thanks for helping <3 keep it up

I am glad it works again. :)

Cheers,
Matthias

closed 2022-12-08 by MatthiasKerbl