AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #852: sailing mode isnot work

lambs13, 2022-12-14:


lambs13, 2022-12-14:
I turned on the sailing mode, and the settings are as shown in the attached picture, but the battery is always at 80%, and it will not sail to 60%.
[image]
[image]

MatthiasKerbl, 2022-12-14:
Hi @lambs13,

The Sailing Mode is working as intended. It is not designed to discharge your battery and it can take weeks for it to fall back to the lower end. Please read more about what the Sailing Mode does in our blog post here: https://apphousekitchen.com/feature-explanation-sailing-mode/

Cheers,
Matthias

closed 2022-12-14 by MatthiasKerbl