AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #855: aldente not working after updating to macOs Ventura 13.1

zynth17, 2022-12-15:
**Describe the bug**
after updating to macOS ventura 13.1 the discharge function is not working. it still charging even after the charge limit

**To Reproduce**
Steps to reproduce the behavior:
1. update to macOS ventura 13.1
2. start charging until the limit
3. See error

**Expected behavior**
stop the charge after reach the limit

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - M1 Pro 14"
 - macOS version Ventura 13.1
 - AlDente Pro 
 - AlDente Version 1.19.2

MatthiasKerbl, 2022-12-15:
Hi @zynth17,

This might be due to a common bug after a macOS update.
Please try the following steps to fix it and you can read more about it here #768.

> AlDente stops working after the macOS update!
> 
> Unfortunately, AlDente temporarily does not work after a macOS update for a small number of users.
> 
> To get AlDente to work again, please try the following steps which have worked for many users before:
> 
> 1. Unplug and discharge your MacBook to less than your current charge limit in AlDente
> 2. Turn off your MacBook and close the lid for a couple of minutes
> 3. Unplug the power brick from the outlet and wait a couple of minutes
> 4. Turn on your MacBook and plug it in again
> 5. Check if it stops charging at the charge limit set in AlDente
> 
> We are aware of this bug for a while now but were so far not able to fix it. However, we will keep trying.

Cheers,
Matthias

MatthiasKerbl, 2022-12-27:
Hi @zynth17,

Could you get it to work again with the steps mentioned above?

Cheers,
Matthias

sandr8, 2023-01-01:
IIRC, for me reinstalling the helper fixes it every time. Will come back here to let you know after my next upgrade.

MatthiasKerbl, 2023-01-12:
Hi @zynth17,

I will close this issue for now since you didn't respond in a while. However, feel free to come back if you still experience this issue.

Cheers,
Matthias

closed 2023-01-12 by MatthiasKerbl