AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #857: License problem

Omairfp, 2022-12-20:
Hi, I pruchased aldente pro and its been several hours but didn't get my license key yet...what to do? 
Email - omairfp@gmail.com
Card ending with 6165

Omairfp, 2022-12-20:
No contact e-mail is available from your website. I have tried to reach by e-mail at help@paddle.com but this is not very helpful. What is it? Am I supposed to pay and forget... is it a scam

Bachsau, 2022-12-22:
Most likely you just have a crappy spam filter. Paddle sends you the license key via email shortly after your payment.

MatthiasKerbl, 2022-12-22:
Hi @Omairfp,

I am sorry you are having trouble with the buying process.
Please take a look at the following steps from our FAQ:

**I Did Not Get The License Code Or Any Receipt Emails After Buying AlDente Pro!**

There are multiple reasons why you did not get your license code after the purchase:

1. Sometimes it takes up to 30 Minutes for the purchase emails to go through.
2. The purchase emails end up in the junk email folder.
3. You had a typo in the email address during your purchase.
4. Failures of the payment networks/providers.

Please contact the support team of our payment provider Paddle at [assist@paddle.com](mailto:assist@paddle.com) and include the email address you used to purchase AlDente Pro. The Paddle support will try to solve your issue as fast as possible.

Cheers,
Matthias

MatthiasKerbl, 2023-01-12:
Hi @Omairfp,

Could you solve this issue with the help of Paddle support?

Cheers,
Matthias

closed 2023-03-06 by MatthiasKerbl