AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #858: Scripts on power/battery

dgancedo, 2022-12-20:
I think it would be great to have support to run scrips on power/battery events.
In example run an script when AC Adapter its connected or removed can help to open/close apps automatically when you connect to an external power source

For example I use some Intel Apps on my M1 mac and sometimes I forgot to close it when I leave then this apps drain the battery really quickly, and script that close this apps when I'm on battery and open again in AC power could help to solve this.

MatthiasKerbl, 2022-12-24:
Hi @dgancedo,

Adding AppleScripts support to AlDente definitely makes sense. I will add it to our list for future updates.
For now, maybe you can add some of the functionality you are looking for with Apple Shortcuts: https://apphousekitchen.com/feature-explanation-shortcuts/

Cheers,
Matthias

MatthiasKerbl, 2022-12-26:
Hi @dgancedo,

I just saw that we already have an issue open for this feature request here: #638
Therefore I will close this issue but feel free to add more ideas to the other thread.

Cheers,
Matthias

closed 2022-12-26 by MatthiasKerbl