AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #864: Customizable Magsafe LED behavior?

sandr8, 2023-01-01:
I really like the newest pro feature that takes control of the LED on the magsafe connector.

Wondering if it would be possible to invert the meaning of amber and red, such that the LED would only be bright red when actually charging and most of the time it would be amber. Asking because I hate all blue/green light at night :-)

Or I'm even wondering if it would be possible to do something more sophisticated. Perhaps something like:

```
                 amber - I'm charging
              no light - great! I'm just running from the mains without charging
 slowly blinking green - draining battery, need a more beefy power brick with this workload
rapidly blinking green - no power: are you sure the brick on the other end is plugged into the wall socket?
```

I'm hoping that the middle pin can still work even if the two neutral pins are only connected to the laptop, and that the necessary low-level access would be available. Otherwise, sorry for wasting your time with my little pipe dream :-)

sandr8, 2023-01-11:
Looking a little bit into things, I figured something like that wouldn't be possible. I personally don't like having a green light disrupting my sleep at night, but would like to be warned if the battery is accidentally charging beyond my preset limit.

How about an option to automatically set the green light threshold just a little bit above the charging limit?

MatthiasKerbl, 2023-01-12:
> I really like the newest pro feature that takes control of the LED on the magsafe connector.
>
> Wondering if it would be possible to invert the meaning of amber and red, such that the LED would only [...]

Hi @sandr8,

Inverting the meaning of the colors would be possible. However, in our opinion, it makes the most sense and is the least confusing to most people when it is done the way we implemented it.

Unfortunately, it is not possible to implement more advanced features such as fast and slow blinking, since it is highly limited what we can control about the LED.

> How about an option to automatically set the green light threshold just a little bit above the charging limit?

This would be possible, however, during normal usage, the MacBook should not charge above the Charge Limit. Therefore, a threshold like that is not necessary in my opinion.

Cheers,
Matthias

sandr8, 2023-01-13:
> Inverting the meaning of the colors would be possible.

There seem to be other people who share my same (maybe paranoid) concern:

https://superuser.com/questions/839652/how-do-i-turn-off-magsafe-light

> However, in our opinion, it makes the most sense and is the least confusing to most people when it is done the way we implemented it.

If you are worried about the complexity of the code and its maintability, I understand. If you are worried about the user experience and the complexity of the UI, maybe this could be hidden under the hood via a text-based configuration file?

MatthiasKerbl, 2023-01-13:
Hi @sandr8,

> There seem to be other people who share my same (maybe paranoid) concern:
> https://superuser.com/questions/839652/how-do-i-turn-off-magsafe-light

Offering a feature to completely turn off the MagSafe LED definitely makes a lot of sense in our opinion. That's why we implemented the option to turn off the LED for MagSafe 3 MacBooks. Unfortunately, we did not find a possibility to offer this option for MagSafe 2 MacBooks.

Adding an option to invert the color of the LED is a different story. We discussed it internally but decided that it does not make too much sense to add this feature. Furthermore, it also adds usability, code, and maintenance complexity. However, if a lot of users want it, we might add it in the future.

Cheers,
Matthias

jconley, 2023-01-23:
personally, i would like to see orange when charging, but off when not charging.

closed 2024-09-03 by MatthiasKerbl